ros-industrial-attic/bezier #4: Unexpected "normal markers"

FrancoisLasson, 2015-06-30:
In Rviz, we create and display markers for meshes, trajectories on mesh, extrication paths and normals.

For the normal markers, we generate a `MarkerArray` for each pass. After a few passes, we could see several unexpected markers (Cf out of pass markers on pictures below) 

I think that this problem could be due to : 
-`MarkerArray` refresh problem
-Our `Bezier::displayNormal` function
-Our `Bezier::generateTrajectory` function that calling Bezier::displayNormal

[image]
[image]

gavanderhoorn, 2015-06-30:
The 'unexpected markers' are the markers 'in the air', free-floating above the surface?

FrancoisLasson, 2015-06-30:
Exactly, the markers outside the trajectory (green line)

PR #10 (2015-07-07): Fix old normals not being deleted when visualizing new data
Fixes #4 
@FrancoisLasson needs review + testing
closed 2015-07-07 by FrancoisLasson