ros-industrial-attic/bezier #12: VTK 6.3 warning: UseVTK.cmake is not a directory

VictorLamoine, 2015-07-08:
When compiling:
```bash
cc1plus: warning: /usr/local/lib/cmake/vtk-6.3/UseVTK.cmake: not a directory [enabled by default]
```

This is probably VTK related.

ehsan-asadi, 2015-10-01:
Hi Victor,
I got a similar compiling error. I am wondering if you could handle the problem and how? Thank you in advance.

ehsan-asadi, 2015-10-01:
After that warning, the process failed and I got this error; 
No rule to make target `/home/work_space/VTK/build/lib/libvtkFiltersHybrid-6.3.so.1' needed by `/home/catkin_ws/devel/lib/bezier_application/bezier_application

VictorLamoine, 2015-10-01:
Can you share your `CMakeLists.txt` configuration of VTK?
I'll take a look at this in the week.

ehsan-asadi, 2015-10-03:
Thanks a lot Victor. I would appreciate your help.
I rebuilt the VTK using the given configuration in above and now I am running to a similar warning but with different errors.

cc1plus: warning: /home/work_space/VTK/CMake/UseVTK.cmake: not a directory [enabled by default]
/home/catkin_ws/src/bezier/bezier_library/src/bezier_library.cpp: In member function ‘bool Bezier::generateStripperOnSurface(vtkSmartPointer<vtkPolyData>, std::vector<std::vector<std::pair<Eigen::Matrix<double, 3, 1>, Eigen::Matrix<double, 3, 1> > > >&)’:
/home/catkin_ws/src/bezier/bezier_library/src/bezier_library.cpp:556:13: error: ‘class vtkStripper’ has no member named ‘SetJoinContiguousSegments’
   stripper->SetJoinContiguousSegments(true);
             ^
/home/catkin_ws/src/bezier/bezier_library/src/bezier_library.cpp: In member function ‘bool Bezier::generateTrajectory(std::vector<Eigen::Transform<double, 3, 2>, Eigen::aligned_allocator<Eigen::Transform<double, 3, 2> > >&, std::vector<bool>&, std::vector<int>&)’:
/home/catkin_ws/src/bezier/bezier_library/src/bezier_library.cpp:844:27: error: ‘class vtkStripper’ has no member named ‘SetJoinContiguousSegments’
     extrication_stripper->SetJoinContiguousSegments(true);

ehsan-asadi, 2015-10-03:
make[3]: *** [bezier/bezier_library/CMakeFiles/bezier_library_obj.dir/src/bezier_library.cpp.o] Error 1
make[2]: *** [bezier/bezier_library/CMakeFiles/bezier_library_obj.dir/all] Error 2
make[1]: *** [bezier/bezier_application/CMakeFiles/bezier_application.dir/rule] Error 2
make: *** [bezier_application] Error 2

VictorLamoine, 2015-10-03:
I've updated the instructions on the [README.md](../blob/master/README.md).
My changes were not included in the 6.3 release, they will probably be in 7.0, two options for you:

- Cherry-pick the commits with my improvements,
https://github.com/Kitware/VTK/commit/2b5792f7746fecaf5cb04fe88eca08017e526727, https://github.com/Kitware/VTK/commit/4166cd1d97f6ba9323d4a5e934652cd4db5f641d, https://github.com/Kitware/VTK/commit/136c17f104b6ed9041adb9bb8bef9945942abb74
- Use master branch instead

Then compile and install VTK again, the compilation error will be gone!

ehsan-asadi, 2015-10-03:
After trying both suggested solutions in above, now I get the same error as the first time.
It seems something is wrong with vtkFiltersHybrid. I have double checked the VTK configurations and they are the same as yours.
Thanks again. 

No rule to make target /home/work_space/VTK/build/lib/libvtkFiltersHybrid-6.3.so.1' needed by/home/catkin_ws/devel/lib/bezier_application/bezier_application

VictorLamoine, 2015-10-05:
Seems like you're trying to compile VTK with `OpenGL2` support instead of `OpenGL`;
` VTK_RENDERING_BACKEND            OpenGL `

It should work with OpenGL2 but I think the CMakeLists.txt has to be modified.

ehsan-asadi, 2015-10-05:
Good Point. I have slightly modified the CMakeLists.txt of bezier_application and now the previous errors are gone. However, still I am not able to run the package.
My CMakeLists.txt; (I am not sure if it is the best way to do that)

cmake_minimum_required(VERSION 2.8.3)
project(bezier_application)

find_package(catkin REQUIRED COMPONENTS roscpp bezier_library tf tf_conversions moveit_ros_planning_interface visualization_msgs)
find_package(PCL 1.8.0 REQUIRED)
find_package(VTK 6.3 REQUIRED COMPONENTS vtkFiltersHybrid NO_MODULE)
catkin_package(CATKIN_DEPENDS bezier_library)
include_directories(${VTK_USE_FILE} ${PCL_INCLUDE_DIRS} ${catkin_INCLUDE_DIRS}) 
add_definitions(${PCL_DEFINITIONS} ${VTK_DEFINITIONS})

if(TARGET vtkRenderingOpenGL2)
  message(STATUS "Building optional volume renderingOpenGL2 component")
  find_package(VTK 6.3 COMPONENTS vtkFiltersHybrid vtkRenderingOpenGL2 vtkRenderingFreeType vtkIOMPIParallel vtkIOMPIImage vtkInteractionStyle NO_MODULE)
endif()

add_executable (bezier_application src/bezier_application.cpp)
if(VTK_LIBRARIES) # May never happen because we expect VTK 6.3 minimum
  target_link_libraries (bezier_application ${catkin_LIBRARIES} ${VTK_LIBRARIES} ${PCL_LIBRARIES} bezier_library)
else()
  target_link_libraries (bezier_application ${catkin_LIBRARIES} vtkHybrid  vtkWidgets ${PCL_LIBRARIES} bezier_library)
endif()

ehsan-asadi, 2015-10-05:
Now, after running the roslaunch, everything start fine but the bezier_application is died immediately.  

core service [/rosout] found
process[industrial_robot_simulator-1]: started with pid [10959]
process[joint_trajectory_action-2]: started with pid [10962]
process[robot_state_publisher-3]: started with pid [11016]
process[move_group-4]: started with pid [11057]
[ INFO] [1444049513.290203910]: Loading robot model 'fanuc_m10ia'...
process[bezier_application-5]: started with pid [11088]
process[rviz-6]: started with pid [11141]
[ INFO] [1444049513.629489344]: joint_1 -3.14 3.14 1
[ INFO] [1444049513.629589077]: joint_2 -1.57 2.79 1
[ INFO] [1444049513.629636358]: joint_3 -3.14 4.61 1
[ INFO] [1444049513.629684166]: joint_4 -3.31 3.31 1
[ INFO] [1444049513.629732994]: joint_5 -3.31 3.31 1
[ INFO] [1444049513.629779201]: joint_6 -6.28 6.28 1
Fontconfig error: "misc.conf", line 41: invalid attribute 'name'
Fontconfig error: "misc.conf", line 50: invalid attribute 'name'
Fontconfig error: "misc.conf", line 59: invalid attribute 'name'
Fontconfig error: "misc.conf", line 70: invalid attribute 'name'
[ INFO] [1444049513.708394770]: Publishing maintained planning scene on 'monitored_planning_scene'
[ INFO] [1444049513.714378208]: MoveGroup debug mode is OFF
Starting context monitors...
[ INFO] [1444049513.714484349]: Starting scene monitor
[ INFO] [1444049513.721858839]: Listening to '/planning_scene'
[ INFO] [1444049513.721952293]: Starting world geometry monitor
[ INFO] [1444049513.730013436]: Listening to '/collision_object' using message notifier with target frame '/base_link '
[ INFO] [1444049513.740265925]: Listening to '/planning_scene_world' for planning scene world geometry
[ INFO] [1444049513.789620427]: Listening to '/attached_collision_object' for attached collision objects
Context monitors started.
[ INFO] [1444049513.942063234]: Initializing OMPL interface using ROS parameters
[ INFO] [1444049513.972738394]: Using planning interface 'OMPL'
[ INFO] [1444049514.106793943]: Param 'default_workspace_bounds' was not set. Using default value: 10
[ INFO] [1444049514.108010477]: Param 'start_state_max_bounds_error' was set to 0.1
[ INFO] [1444049514.108980132]: Param 'start_state_max_dt' was not set. Using default value: 0.5
[ INFO] [1444049514.110022905]: Param 'start_state_max_dt' was not set. Using default value: 0.5
[ INFO] [1444049514.110901399]: Param 'jiggle_fraction' was set to 0.05
[ INFO] [1444049514.111977003]: Param 'max_sampling_attempts' was not set. Using default value: 100
[ INFO] [1444049514.112104743]: Using planning request adapter 'Add Time Parameterization'
[ INFO] [1444049514.112152863]: Using planning request adapter 'Fix Workspace Bounds'
[ INFO] [1444049514.112192997]: Using planning request adapter 'Fix Start State Bounds'
[ INFO] [1444049514.112234541]: Using planning request adapter 'Fix Start State In Collision'
[ INFO] [1444049514.112277167]: Using planning request adapter 'Fix Start State Path Constraints'
===============================================================================REQUIRED process [bezier_application-5] has died!
process has died [pid 11088, exit code -11, cmd /home/ehsan/catkin_ws/devel/lib/bezier_application/bezier_application __name:=bezier_application __log:=/home/ehsan/.ros/log/2b00200e-6b5e-11e5-87d5-d8d3853c039a/bezier_application-5.log].
log file: /home/ehsan/.ros/log/2b00200e-6b5e-11e5-87d5-d8d3853c039a/bezier_application-5*.log
Initiating shutdown!
===============================================================================

VictorLamoine, 2015-10-05:
You should have provided the logs :wink: 

I think that PCL has to be compiled/installed against the same VTK version you're using to build the Bezier library otherwise they'll be clashes.
As PCL cannot be compiled against VTK using OpenGL2 (see https://github.com/PointCloudLibrary/pcl/issues/1275) I'd suggest compiling with OpenGL for the moment; I'm not sure it's ever going to work with OpenGL2 as is.

ehsan-asadi, 2015-10-06:
I was worried about having my PCL compiled against different version of VTK, but I was hopping that it may work this way since I use other packages depending on the PCL and they work fine. Please find attached the logs.

On the other note, for installing PCL and VTK if it is necessary to downgrade my OpneGL2, already have, to OpenGL  or it is only needed to set the VTK  configuration VTK_RENDERING_BACKEND = OpenGL.

VictorLamoine, 2015-10-06:
- Set `VTK_RENDERING_BACKEND` to `OpenGL` in VTK CMake configuration
Compile / install VTK

- Configure PCL to compile against the previously installed VTK (`/usr/local/lib/cmake/vtk-6.3` by default)
Compile / install PCL

Then you will be able to build and run the Bezier library fine.
I tried to make PCL work with OpenGL but it will require some work before it does; I'll keep you posted when it does.

ehsan-asadi, 2015-10-06:
Thanks Victor. I will follow your instruction.

VictorLamoine, 2015-10-08:
@ehsan-asadi Did you get it working?

ehsan-asadi, 2015-10-08:
Tomorrow, I am going to reinstall the VTK and PCL and I will let you know about the result.

ehsan-asadi, 2015-10-10:
I'm having problem compiling PCL (master, clone from github) against the VTK (master, clone from github) lately. I get a file not found for  vtkIOImageModule.h 
fatal error: vtkIOImageModule.h: No such file or directory  #include "vtkIOImageModule.h"

ehsan-asadi, 2015-10-10:
After changing the VTK path from /user/local to my /home/.../VTK/build address, finally I have managed to compile the PCL and Bezier against VTK 6.3. Now the package works fine. I am wondering if there is any instruction or document about the Bezier more than the given information in the Github. Thanks Victor.

VictorLamoine, 2015-10-10:
Great!

No there is nothing more than what's on the repository (see [doc/REAMDE.md](https://github.com/ros-industrial-consortium/bezier/blob/master/bezier_library/doc/README.md]))
You can also generate the doxygen documentation of the API with the CMake target:`bezier_library_doc` ... and ask me questions :wink:

shreyasgan, 2018-07-03:
Hello, 
@VictorLamoine 
I have this same problem, but with VTK 7.1 and PCL 1.8 in Ubuntu 16.04. When I compiled and installed the required files from the README.txt, and roslaunched the launch file, the bezier-surfacing-application would die, and the UseVTK.cmake error would show when I catkin_make my catkin_ws. After applying these changes, I get this error while trying to compile PCL against VTK 7.1 with OpenGL

/usr/bin/ld: cannot find -lvtkRenderingGL2PSOpenGL2
/usr/bin/ld: cannot find -lvtkRenderingOpenGL2
/usr/bin/ld: cannot find -lvtkglew
/usr/bin/ld: cannot find -lvtkRenderingContextOpenGL2
/usr/bin/ld: cannot find -lvtkDomainsChemistryOpenGL2
/usr/bin/ld: cannot find -lvtkRenderingVolumeOpenGL2
collect2: error: ld returned 1 exit status
io/CMakeFiles/pcl_io.dir/build.make:797: recipe for target 'lib/libpcl_io.so.1.8.1.99' failed
make[2]: *** [lib/libpcl_io.so.1.8.1.99] Error 1
CMakeFiles/Makefile2:287: recipe for target 'io/CMakeFiles/pcl_io.dir/all' failed
make[1]: *** [io/CMakeFiles/pcl_io.dir/all] Error 2

Any help would be greatly appreciated.

VictorLamoine, 2018-07-03:
Try after a `sudo ldconfig`

shreyasgan, 2018-07-05:
Thank you for the timely response, 

After running sudo ldconfig after compiling and installing VTK under this configuration, I am still getting the same error when compiling PCL.

BUILD_DOCUMENTATION              OFF                                          
 BUILD_EXAMPLES                   OFF                                          
 BUILD_SHARED_LIBS                ON                                           
 BUILD_TESTING                    ON                                           
 CMAKE_BACKWARDS_COMPATIBILITY    2.4                                          
 CMAKE_BUILD_TYPE                 Release                                      
 CMAKE_INSTALL_PREFIX             /usr/local                                   
 CTEST_TEST_TIMEOUT               3600                                         
 EXECUTABLE_OUTPUT_PATH                                                        
 LIBRARY_OUTPUT_PATH                                                           
 VTK_ANDROID_BUILD                OFF                                          
 VTK_EXTRA_COMPILER_WARNINGS      OFF                                          
 VTK_Group_Imaging                ON                                           
 VTK_Group_MPI                    ON                                           
 VTK_Group_Qt                     ON                                           
 VTK_Group_Rendering              ON        
VTK_Group_StandAlone             ON                                           
 VTK_Group_Tk                     ON                                           
 VTK_Group_Views                  ON                                           
 VTK_Group_Web                    OFF                                          
 VTK_IOS_BUILD                    OFF                                          
 VTK_PYTHON_VERSION               2                                            
 VTK_QT_VERSION                   4                                            
 VTK_RENDERING_BACKEND            OpenGL                                       
 VTK_SMP_IMPLEMENTATION_TYPE      Sequential                                   
 VTK_USE_CXX11_FEATURES           OFF                                          
 VTK_USE_LARGE_DATA               OFF                                          
 VTK_WRAP_JAVA                    OFF                                          
 VTK_WRAP_PYTHON                  OFF                                          
 VTK_WRAP_TCL                     OFF

VictorLamoine, 2018-07-06:
Ok let's try with a newer version of VTK and PCL from sources.

I have tested these instructions on Ubuntu 16.04 and Kubuntu 18.04.

# Visualisation ToolKit
```bash
sudo apt install -y libglew-dev libxt-dev libqt5x11extras5
mkdir -p $HOME/libraries/VTK-8.1/build_release
cd $HOME/libraries/VTK-8.1/
wget http://www.vtk.org/files/release/8.1/VTK-8.1.0.zip
unzip VTK-8.1.0.zip
mv VTK-8.1.0 src
cd build_release
cmake ../src -DCMAKE_BUILD_TYPE=Release
make -j4
sudo make -j4 install
sudo ldconfig
```

# Point Cloud Library
```
mkdir -p $HOME/libraries/pcl/build_release
cd $HOME/libraries/pcl/
git clone https://github.com/PointCloudLibrary/pcl.git src
cd build_release
cmake ../src
make -j4
sudo make -j4 install
```

shreyasgan, 2018-07-09:
VTK and PCL were compiled and installed correctly, but when I clone and compile the bezier project, this error comes up:

CMake Error at bezier/bezier_library/CMakeLists.txt:15 (find_package):
  Could not find a configuration file for package "VTK" that is compatible
  with requested version "8.0.0".

  The following configuration files were considered but not accepted:

    /usr/lib/cmake/vtk-6.2/VTKConfig.cmake, version: 6.2.0

-- Configuring incomplete, errors occurred!
See also "/home/shreyas/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/shreyas/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

How can I resolve this error?

VictorLamoine, 2018-07-10:
It seems like VTK is not installed properly beceause CMake doesn't find it.

What is the output of sudo `find / -name "VTKConfig.cmake"` ?

On my machine:
```
/home/victor/libraries/VTK-8.1.1/build_release/VTKConfig.cmake
/home/victor/libraries/VTK-8.1.1/build_release/CMakeFiles/VTKConfig.cmake
find: ‘/run/user/1000/gvfs’: Permission non accordée
/usr/lib/cmake/vtk-6.3/VTKConfig.cmake
/usr/local/lib/cmake/vtk-8.1/VTKConfig.cmake
```

You should have the last line in your result

shreyasgan, 2018-07-10:
I ran sudo make install inside the build release directory of VTK, ran find / -name "VTKConfig.cmake" and this was the result.

/usr/lib/cmake/vtk-6.2/VTKConfig.cmake
/usr/local/lib/cmake/vtk-8.1/VTKConfig.cmake
/home/shreyas/libraries/VTK-8.1/build_release/VTKConfig.cmake
/home/shreyas/libraries/VTK-8.1/build_release/CMakeFiles/VTKConfig.cmake
find: ‘/run/user/1000/gvfs’: Permission denied
/media/shreyas/b7964f5c-019f-4833-b7e0-44a4f60e5dfc/libraries/VTK-7.1/lib/cmake/vtk-7.1/VTKConfig.cmake
/media/shreyas/b7964f5c-019f-4833-b7e0-44a4f60e5dfc/shreyas/libraries/VTK-7.1/src/CMakeFiles/VTKConfig.cmake
/media/shreyas/b7964f5c-019f-4833-b7e0-44a4f60e5dfc/shreyas/libraries/VTK-7.1/src/VTKConfig.cmake
/media/shreyas/b7964f5c-019f-4833-b7e0-44a4f60e5dfc/shreyas/libraries/VTK-7.1/build_release/CMakeFiles/VTKConfig.cmake
/media/shreyas/b7964f5c-019f-4833-b7e0-44a4f60e5dfc/shreyas/libraries/VTK-7.1/build_release/VTKConfig.cmake

After that, I ran catkin_make inside my catkin_ws and it successfully compiled, but I got this warning near the end of it

[ 84%] Building CXX object bezier/bezier_library/CMakeFiles/bezier_library.dir/src/bezier_library.cpp.o
cc1plus: warning: /usr/local/lib/cmake/vtk-8.1/UseVTK.cmake: not a directory
[ 84%] Built target rviz_visual_tools_demo
[ 85%] Linking CXX executable /home/shreyas/catkin_ws/devel/lib/rviz_visual_tools/imarker_simple_demo
[ 85%] Built target imarker_simple_demo
[ 87%] Linking CXX shared library /home/shreyas/catkin_ws/devel/lib/libbezier_library.so
[ 87%] Built target bezier_library
Scanning dependencies of target bezier_library_tire_machining
Scanning dependencies of target bezier_library_surfacing
Scanning dependencies of target bezier_library_painting
[ 88%] Building CXX object bezier/bezier_library/CMakeFiles/bezier_library_painting.dir/src/bezier_painting.cpp.o
cc1plus: warning: /usr/local/lib/cmake/vtk-8.1/UseVTK.cmake: not a directory
[ 89%] Building CXX object bezier/bezier_library/CMakeFiles/bezier_library_surfacing.dir/src/bezier_surfacing.cpp.o
[ 90%] Building CXX object bezier/bezier_library/CMakeFiles/bezier_library_tire_machining.dir/src/bezier_tire_machining.cpp.o
cc1plus: warning: /usr/local/lib/cmake/vtk-8.1/UseVTK.cmake: not a directory
cc1plus: warning: /usr/local/lib/cmake/vtk-8.1/UseVTK.cmake: not a directory
[ 92%] Linking CXX shared library /home/shreyas/catkin_ws/devel/lib/libbezier_library_tire_machining.so
[ 93%] Linking CXX shared library /home/shreyas/catkin_ws/devel/lib/libbezier_library_painting.so

I believe I was getting this error earlier, so I ran the fanuc_m20ia_surfacing.launch file and the application died on me.

VictorLamoine, 2018-07-11:
The compilation warning is ok.
Now you need to compile the ROS workspace in debug mode and determine what is crashing. See http://wiki.ros.org/roslaunch/Tutorials/Roslaunch%20Nodes%20in%20Valgrind%20or%20GDB

With a gdb trace you should be able to understand what's wrong. Check the issues on the repository you will probably find similar cases.

I do not have time to maintain this repository so it is not up to date and there are bugs.