ros-industrial-attic/bezier #13: Misclaneous improvments

VictorLamoine, 2015-11-26:
To be done by the trainee in 2016 before trying to improve anything:

- [ ] Fix code style (some little errors here and there)
- [ ] Document the project clearly, images etc. Explain in detail the parameters:
`grind_depth_, effector_diameter_, covering_, extrication_coefficient_, extrication_frequency_`
- [ ] Simplify `bezier_application` package.
- [ ] Rename variables:
 - `grind_depth_` > `maximum_depth_of_pass_`
 - `grind_depth_` > `maximum_depth_of_pass_`
 - `covering_` > `covering_percentage_`
- [ ] Allow providing a lean angle for the pose generation (axis / angle).
- [ ] Remove `this->` whenever possible to simplify code.
- [ ] Improve error messages (eg: print file name here [bezier_library.cpp#L29](https://github.com/ros-industrial-consortium/bezier/blob/master/bezier_library/src/bezier_library.cpp#L29)).

KevinBollore, 2016-02-09:
- [ ] Improve Doxygen documentation (some functions are not documented)

VictorLamoine, 2016-02-16:
Fixed with #17

PR #15 (2016-02-16): Misclaneous improvements
Partially fixes #13
closed 2016-02-16 by VictorLamoine