ros-industrial-attic/bezier #16: Rename `extrication` paths

VictorLamoine, 2016-02-10:
I'm pretty sure "extrication paths" does not mean anything.

@shaun-edwards How would you call a trajectory that is not machining when you just want to move the tool from one point to another.

This corresponds to the red trajectory on the image:
[image]

The green trajectories are removing material from the object.
The red trajectories are just moving the robot between green trajectories.

Is there a specific/technical term for that?

VictorLamoine, 2016-02-15:
@gavanderhoorn An idea?

gavanderhoorn, 2016-02-15:
No, not immediately. I've heard ppl refer to them as *non-toolpath motion*, or *freespace motion*, but I'm pretty sure that that isn't official terminology.

gavanderhoorn, 2016-02-15:
Btw

> I'm pretty sure "extrication paths" does not mean anything.

that's not entirely true: see [wikipedia/Vehicle extrication](https://en.wikipedia.org/wiki/Vehicle_extrication) fi.

The term is also used to describe motions that a robot uses to get out of a certain situation, fi in rescue robotics ([here](https://books.google.nl/books?id=Xpgi5gSuBxsC&pg=PA1168&lpg=PA1168&dq=extrication+path+robot&source=bl&ots=lUllW4lc8T&sig=ECZbwynnDu6Q6e1yuB3YaYdWcTc&hl=en&sa=X&ved=0ahUKEwjYzrukuPnKAhWFNJoKHROkCXkQ6AEILTAD#v=onepage&q&f=false) fi).

VictorLamoine, 2019-04-01:
I don't have time for a renaming.

closed 2019-04-01 by VictorLamoine