ros-industrial-attic/bezier #78: JacobiSVD error

VictorLamoine, 2016-06-20:
roslaunch bezier_application bezier_application_r1000ia.launch meshname:=plane/plane.ply

```
bezier_application: /usr/include/eigen3/Eigen/src/SVD/JacobiSVD.h:632: const MatrixVType& Eigen::JacobiSVD<MatrixType, QRPreconditioner>::matrixV() const [with _MatrixType = Eigen::Matrix<double, 3, 3>; int QRPreconditioner = 2; Eigen::JacobiSVD<MatrixType, QRPreconditioner>::MatrixVType = Eigen::Matrix<double, 3, 3>; typename _MatrixType::Scalar = double]: Assertion `computeV() && "This JacobiSVD decomposition didn't compute V. Did you ask for it?"' failed.
```

VictorLamoine, 2016-06-23:
It only crashes on one machine, all of them are 64 bits, Ubuntu 14.04 up to date.

Victor desktop
```
$ dpkg -l libeigen3-dev
Souhait=inconnU/Installé/suppRimé/Purgé/H=à garder
| État=Non/Installé/fichier-Config/dépaqUeté/échec-conFig/H=semi-installé/W=attend-traitement-déclenchements
|/ Err?=(aucune)/besoin Réinstallation (État,Err: majuscule=mauvais)
||/ Nom                                           Version                     Architecture                Description
+++-=============================================-===========================-===========================-===============================================================================================
ii  libeigen3-dev                                 3.2.5-4~ubuntu14.04.1~ppa1  all                         lightweight C++ template library for linear algebra```

@Nandite & @KevinBollore can you post the output of the above command? Thanks!

Nandite, 2016-06-23:
@VictorLamoine 
[image]

KevinBollore, 2016-06-24:
```
dell@M4800:~$ dpkg -l libeigen3-dev
Souhait=inconnU/Installé/suppRimé/Purgé/H=à garder
| État=Non/Installé/fichier-Config/dépaqUeté/échec-conFig/H=semi-installé/W=attend-traitement-déclenchements
|/ Err?=(aucune)/besoin Réinstallation (État,Err: majuscule=mauvais)
||/ Nom                     Version          Architecture     Description
+++-=======================-================-================-====================================================
ii  libeigen3-dev           3.2.0-8          all              lightweight C++ template library for linear algebra
```

Nandite, 2016-06-24:
@VictorLamoine 
From the right computer this time.
[image]

VictorLamoine, 2016-06-28:
Also see https://github.com/davetcoleman/rviz_visual_tools/issues/36

gavanderhoorn, 2016-07-01:
To add a data point, I've got `libeigen3-dev` version `v3.2.5-4~ubuntu14.04.1~ppa1`.

VictorLamoine, 2016-07-03:
Backtrace:
```
Backtrace
```
(gdb) bt
#0  0x00007ffff4719c37 in __GI_raise (sig=sig@entry=6) at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
#1  0x00007ffff471d028 in __GI_abort () at abort.c:89
#2  0x00007ffff4712bf6 in __assert_fail_base (fmt=0x7ffff48633b8 "%s%s%s:%u: %s%sAssertion `%s' failed.\n%n", 
    assertion=assertion@entry=0x7ffff1f3d690 "computeV() && \"This JacobiSVD decomposition didn't compute V. Did you ask for it?\"", 
    file=file@entry=0x7ffff1f3d628 "/usr/include/eigen3/Eigen/src/SVD/JacobiSVD.h", line=line@entry=632, 
    function=function@entry=0x7ffff1f48800 <Eigen::JacobiSVD<Eigen::Matrix<double, 3, 3, 0, 3, 3>, 2>::matrixV() const::__PRETTY_FUNCTION__> "const MatrixVType& Eigen::JacobiSVD<MatrixType, QRPreconditioner>::matrixV() const [with _MatrixType = Eigen::Matrix<double, 3, 3>; int QRPreconditioner = 2; Eigen::JacobiSVD<MatrixType, QRPreconditio"...) at assert.c:92
#3  0x00007ffff4712ca2 in __GI___assert_fail (assertion=0x7ffff1f3d690 "computeV() && \"This JacobiSVD decomposition didn't compute V. Did you ask for it?\"", 
    file=0x7ffff1f3d628 "/usr/include/eigen3/Eigen/src/SVD/JacobiSVD.h", line=632, 
    function=0x7ffff1f48800 <Eigen::JacobiSVD<Eigen::Matrix<double, 3, 3, 0, 3, 3>, 2>::matrixV() const::__PRETTY_FUNCTION__> "const MatrixVType& Eigen::JacobiSVD<MatrixType, QRPreconditioner>::matrixV() const [with _MatrixType = Eigen::Matrix<double, 3, 3>; int QRPreconditioner = 2; Eigen::JacobiSVD<MatrixType, QRPreconditio"...) at assert.c:101
#4  0x00007ffff1f1a90c in Eigen::JacobiSVD<Eigen::Matrix<double, 3, 3, 0, 3, 3>, 2>::matrixV (this=0x7fffffffbdc0) at /usr/include/eigen3/Eigen/src/SVD/JacobiSVD.h:632
#5  0x00007ffff1f1685b in Eigen::Transform<double, 3, 2, 0>::computeRotationScaling<Eigen::Matrix<double, 3, 3, 0, 3, 3>, Eigen::Matrix<double, 3, 3, 0, 3, 3> > (this=0xa20480, rotation=0x7fffffffbfa0, 
    scaling=0x0) at /usr/include/eigen3/Eigen/src/Geometry/Transform.h:1020
#6  0x00007ffff0ef840e in moveit::core::LinkModel::setGeometry(std::vector<boost::shared_ptr<shapes::Shape const>, std::allocator<boost::shared_ptr<shapes::Shape const> > > const&, std::vector<Eigen::Transform<double, 3, 2, 0>, Eigen::aligned_allocator<Eigen::Transform<double, 3, 2, 0> > > const&) () from /opt/ros/indigo/lib/libmoveit_robot_model.so
#7  0x00007ffff0f129e8 in moveit::core::RobotModel::constructLinkModel(urdf::Link const*) () from /opt/ros/indigo/lib/libmoveit_robot_model.so
#8  0x00007ffff0f15a02 in moveit::core::RobotModel::buildRecursive(moveit::core::LinkModel*, urdf::Link const*, srdf::Model const&) () from /opt/ros/indigo/lib/libmoveit_robot_model.so
#9  0x00007ffff0f180ce in moveit::core::RobotModel::buildModel(urdf::ModelInterface const&, srdf::Model const&) () from /opt/ros/indigo/lib/libmoveit_robot_model.so
#10 0x00007ffff0f18755 in moveit::core::RobotModel::RobotModel(boost::shared_ptr<urdf::ModelInterface const> const&, boost::shared_ptr<srdf::Model const> const&) ()
   from /opt/ros/indigo/lib/libmoveit_robot_model.so
#11 0x00007fffeb07ee1d in robot_model_loader::RobotModelLoader::configure(robot_model_loader::RobotModelLoader::Options const&) () from /opt/ros/indigo/lib/libmoveit_robot_model_loader.so
#12 0x00007fffeb07fead in robot_model_loader::RobotModelLoader::RobotModelLoader(robot_model_loader::RobotModelLoader::Options const&) () from /opt/ros/indigo/lib/libmoveit_robot_model_loader.so
#13 0x00007ffff193cde6 in moveit::planning_interface::getSharedRobotModel(std::string const&) () from /opt/ros/indigo/lib/libmoveit_common_planning_interface_objects.so
#14 0x00007ffff71b5257 in moveit::planning_interface::MoveGroup::MoveGroupImpl::MoveGroupImpl(moveit::planning_interface::MoveGroup::Options const&, boost::shared_ptr<tf::Transformer> const&, ros::Duration const&) () from /opt/ros/indigo/lib/libmoveit_move_group_interface.so
#15 0x00007ffff713fa1e in moveit::planning_interface::MoveGroup::MoveGroup(std::string const&, boost::shared_ptr<tf::Transformer> const&, ros::Duration const&) ()
   from /opt/ros/indigo/lib/libmoveit_move_group_interface.so
#16 0x00000000004279d3 in main (argc=1, argv=0x7fffffffda68) at /home/victor/catkin_test/src/bezier/bezier_application/src/bezier_application.cpp:122
```

VictorLamoine, 2016-07-07:
Weird thing is that it works in fanuc_grinding:
https://github.com/InstitutMaupertuis/fanuc_grinding/blob/indigo-devel/path_planning/src/path_planning.cpp#L83-L132

VictorLamoine, 2016-07-08:
Does not happen anymore on my machine, I don't know why.

VictorLamoine, 2016-07-13:
@gavanderhoorn would you be so kind as to test again? Since a ROS update it's not crashing so I think the problem was on Move-It side

gavanderhoorn, 2016-07-13:
@VictorLamoine: which branch(es) should I be using? The same as before (on the Inst M fork), or something else?

Might be a good time to do something about #88 :).

VictorLamoine, 2016-07-13:
This repository is the right one for testing, it is the official/up-to-date repo.
The InstitutMaupertuis fork was just temporary (until we got a working version we kept it there)

gavanderhoorn, 2016-07-13:
Just tested, again a crash with exception:
```
bezier_application: /usr/include/eigen3/Eigen/src/SVD/JacobiSVD.h:632: const MatrixVType& Eigen::JacobiSVD<MatrixType, QRPreconditioner>::matrixV() const [with _MatrixType = Eigen::Matrix<double, 3, 3>; int QRPreconditioner = 2; Eigen::JacobiSVD<MatrixType, QRPreconditioner>::MatrixVType = Eigen::Matrix<double, 3, 3>; typename _MatrixType::Scalar = double]: Assertion `computeV() && "This JacobiSVD decomposition didn't compute V. Did you ask for it?"' failed.
```

Up-to-date with ROS (as of today):

 - RViz: `1.11.14-0trusty-20160628-085352-0700`
 - Eigen: `3.2.5-4~ubuntu14.04.1~ppa1`

amd64, Ubuntu Trusty.

VictorLamoine, 2016-07-13:
Well then I'm not closing.. Thanks for testing!