ros-industrial-attic/bezier #116: bezier_application: missing dependency on 'fanuc_m20ia_moveit_config'?

gavanderhoorn, 2016-07-13:
Just tried to launch `bezier_application bezier_application_m20ia.launch surfacing_mode:=true mesh_cad:=plane/plane_defect.ply`, which fails with a `ResourceNotFound` exception complaining about `fanuc_m20ia_moveit_config`. That makes sense, since I don't have that package installed. 

`rosdep` did not ask me to install it.

Is that dependency left out on purpose? Otherwise it might be nice to add it to the manifest.

VictorLamoine, 2016-07-13:
The dependency is missing and that is an error.

Requiring this will also require `fanuc_m20ia_support` and all other fanuc packages indirectly right?

gavanderhoorn, 2016-07-13:
yes, it will automatically pull in all the dependencies as well.

gavanderhoorn, 2016-07-13:
Not 'all others' btw, just the bits that `fanuc_m20ia_moveit_config` needs (ikfast plugin, driver, support and resources).

gavanderhoorn, 2016-07-13:
Btw: adding a `roslaunch` testing test to your package would automatically discover these things. See the tests in `fanuc_m10ia_support` [here](https://github.com/ros-industrial/fanuc/blob/e632abcb4c8f223879884da56fe412a6d488066a/fanuc_m10ia_support/CMakeLists.txt#L9-L13).

closed 2016-07-19 by VictorLamoine