ros-industrial-attic/bezier #132: Residual lines in vtkStripper

Nandite, 2016-08-01:
@VictorLamoine 
[image]
This problem appeared when the `vtkStripper` representing a line sliced through a mesh contains more than one lines. In this case we have 2 lines. The first line contains the "useful" part (point 0 to 86 on the picture) of the extrication trajectory with 250 points (before the neighbor filtering is applied). The second line which i call 'residual line' contains 2 points (87 and 88 on the picture). This is the second line which brings problem for the following processes (extrication filter by example). Take a look at the terminal output : 
[image]

The solution i offer for now is to remove the line if it has less than a specified number of points.

VictorLamoine, 2016-08-01:
This is very dangerous because if we use a rather large sampling (big distance between neighbors) and parts with holes, we are going to delete smaller lines that should not be deleted!

Nandite, 2016-08-01:
Yeah, sure but what is your definition of small line ? For me a small line is something like less than 3 points. And should we allow generation of poses on line with only 2 points ? By the way for the meshes with holes, i've made a discovery earlier : [vtkFillHolesFilter](http://www.vtk.org/doc/nightly/html/classvtkFillHolesFilter.html) . We can use this filter to fills hole and generate grinding pose on it. It will be as if we didnt have holes on the mesh. And thus resolve the above problem (we then remove residual lines, meaning lines with less than 2 or 3 points). What do you think about that ?

Nandite, 2016-08-02:
@VictorLamoine By the way, we can implement the solution i've offered before (another filter for vtkStripper line size) and use it for now, because we haven't yet included the case when the mesh have holes in the library. What do you think about that ?

PR #145 (2016-08-24): Extrication filter improved
Fix #132 : This was fixed by applying a modification on the extrication filter. Indeed, instead of computing the angle starting from each extremity of extrication path, we start from the middle of the trajectory and we iterate through each direction. This method allows to avoid computing angles with points located in the extremity of the trajectory and which are generated by the residual line in the `vtkStripper`. Those points usually cause the filter fail to find a solution.
closed 2016-08-24 by VictorLamoine