ros-industrial-attic/bezier #137: Smooth polydata

Nandite, 2016-08-03:
@VictorLamoine I've made some experimentation on the polydata used in the library, trying to smooth edges and polys and see if we can get better surface for trajectory generation using [vtkSmoothPolyDataFilter](http://www.vtk.org/doc/nightly/html/classvtkSmoothPolyDataFilter.html). 
This is the results i get : 
[image]
[image]
[image]
[image]

As you can see it, we clearly get better surfaces. What do you think about that ?

VictorLamoine, 2016-08-03:
Doesn't look useful. Switch the shading to a smooth shading and you'll get approximately the same results.

How does this improve the library?

Nandite, 2016-08-03:
It smooth the edges and the boundaries, so maybe it could allow to attenuate the problem of stripper having 2 or more residual lines #132. But forget about that, I have performed tests with that, it does not change the problem.

closed 2016-08-03 by Nandite