ros-industrial-attic/bezier #147: Bezier compiling error

jcgarciaca, 2016-09-05:
Hello,
When I add the bezier repository to my workspace and try to use catkin_make the following errors appear:

```
ctory’ with no type [-fpermissive]
   void displayTrajectory(const EigenSTL::vector_Affine3d &trajectory,
                                                           ^
/home/juliocesar/catkin_ws/src/bezier-indigo-devel/bezier_library/include/bezier_library/bezier_library.hpp:296:50: error: ‘EigenSTL::vector_Affine3d’ has not been declared
                                        EigenSTL::vector_Affine3d &trajectory);
                                                  ^
/home/juliocesar/catkin_ws/src/bezier-indigo-devel/bezier_library/include/bezier_library/bezier_library.hpp:302:37: error: ‘EigenSTL::vector_Affine3d’ has not been declared
   void invertXAxisOfPoses(EigenSTL::vector_Affine3d &line);
                                     ^
/home/juliocesar/catkin_ws/src/bezier-indigo-devel/bezier_library/include/bezier_library/bezier_library.hpp:327:46: error: ‘EigenSTL::vector_Affine3d’ has not been declared
                                    EigenSTL::vector_Affine3d &trajectory);
                                              ^
/home/juliocesar/catkin_ws/src/bezier-indigo-devel/bezier_library/include/bezier_library/bezier_library.hpp:335:43: error: ‘EigenSTL::vector_Affine3d’ has not been declared
   bool harmonizeLineOrientation(EigenSTL::vector_Affine3d &poses,
                                           ^
In file included from /home/juliocesar/catkin_ws/src/bezier-indigo-devel/bezier_library/src/bezier_painting.cpp:1:0:
/home/juliocesar/catkin_ws/src/bezier-indigo-devel/bezier_library/include/bezier_library/bezier_painting.hpp:49:44: error: ‘EigenSTL::vector_Affine3d’ has not been declared
   std::string generateTrajectory(EigenSTL::vector_Affine3d &trajectory,
                                            ^
/home/juliocesar/catkin_ws/src/bezier-indigo-devel/bezier_library/include/bezier_library/bezier_painting.hpp:53:44: error: ‘EigenSTL::vector_Affine3d’ has not been declared
   std::string generateTrajectory(EigenSTL::vector_Affine3d &trajectory,
                                            ^
/home/juliocesar/catkin_ws/src/bezier-indigo-devel/bezier_library/src/bezier_painting.cpp:41:58: error: ‘std::string BezierPainting::generateTrajectory’ is not a static member of ‘class BezierPainting’
 std::string BezierPainting::generateTrajectory(EigenSTL::vector_Affine3d &trajectory,
                                                          ^
/home/juliocesar/catkin_ws/src/bezier-indigo-devel/bezier_library/src/bezier_painting.cpp:41:48: error: ‘vector_Affine3d’ is not a member of ‘EigenSTL’
 std::string BezierPainting::generateTrajectory(EigenSTL::vector_Affine3d &trajectory,
                                                ^
/home/juliocesar/catkin_ws/src/bezier-indigo-devel/bezier_library/src/bezier_painting.cpp:41:75: error: ‘trajectory’ was not declared in this scope
 std::string BezierPainting::generateTrajectory(EigenSTL::vector_Affine3d &trajectory,
                                                                           ^
/home/juliocesar/catkin_ws/src/bezier-indigo-devel/bezier_library/src/bezier_painting.cpp:42:66: error: expected primary-expression before ‘&’ token
                                                std::vector<bool> &is_grinding_pose,
                                                                  ^
/home/juliocesar/catkin_ws/src/bezier-indigo-devel/bezier_library/src/bezier_painting.cpp:42:67: error: ‘is_grinding_pose’ was not declared in this scope
                                                std::vector<bool> &is_grinding_pose,
                                                                   ^
/home/juliocesar/catkin_ws/src/bezier-indigo-devel/bezier_library/src/bezier_painting.cpp:43:48: error: expected primary-expression before ‘const’
                                                const bool display_markers)
                                                ^
/home/juliocesar/catkin_ws/src/bezier-indigo-devel/bezier_library/src/bezier_painting.cpp:44:1: error: expected ‘,’ or ‘;’ before ‘{’ token
 {
 ^
make[2]: *** [bezier-indigo-devel/bezier_library/CMakeFiles/bezier_grinding_surfacing_obj.dir/src/bezier_grinding_surfacing.cpp.o] Error 1
make[1]: *** [bezier-indigo-devel/bezier_library/CMakeFiles/bezier_grinding_surfacing_obj.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
In file included from /home/juliocesar/catkin_ws/src/bezier-indigo-devel/bezier_library/src/bezier_library.cpp:1:0:
/home/juliocesar/catkin_ws/src/bezier-indigo-devel/bezier_library/include/bezier_library/bezier_library.hpp:109:52: error: ‘EigenSTL::vector_Affine3d’ has not been declared
   std::string virtual generateTrajectory(EigenSTL::vector_Affine3d &trajectory,
                                                    ^
/home/juliocesar/catkin_ws/src/bezier-indigo-devel/bezier_library/include/bezier_library/bezier_library.hpp:218:32: error: ‘vector_Affine3d’ in namespace ‘EigenSTL’ does not name a type
   void displayTrajectory(const EigenSTL::vector_Affine3d &trajectory,
                                ^
/home/juliocesar/catkin_ws/src/bezier-indigo-devel/bezier_library/include/bezier_library/bezier_library.hpp:218:59: error: ISO C++ forbids declaration of ‘trajectory’ with no type [-fpermissive]
   void displayTrajectory(const EigenSTL::vector_Affine3d &trajectory,
                                                           ^
/home/juliocesar/catkin_ws/src/bezier-indigo-devel/bezier_library/include/bezier_library/bezier_library.hpp:296:50: error: ‘EigenSTL::vector_Affine3d’ has not been declared
                                        EigenSTL::vector_Affine3d &trajectory);
                                                  ^
/home/juliocesar/catkin_ws/src/bezier-indigo-devel/bezier_library/include/bezier_library/bezier_library.hpp:302:37: error: ‘EigenSTL::vector_Affine3d’ has not been declared
   void invertXAxisOfPoses(EigenSTL::vector_Affine3d &line);
                                     ^
/home/juliocesar/catkin_ws/src/bezier-indigo-devel/bezier_library/include/bezier_library/bezier_library.hpp:327:46: error: ‘EigenSTL::vector_Affine3d’ has not been declared
                                    EigenSTL::vector_Affine3d &trajectory);
                                              ^
/home/juliocesar/catkin_ws/src/bezier-indigo-devel/bezier_library/include/bezier_library/bezier_library.hpp:335:43: error: ‘EigenSTL::vector_Affine3d’ has not been declared
[... 243 lines cut ...]
/home/juliocesar/catkin_ws/src/bezier-indigo-devel/bezier_library/src/bezier_library.cpp:627:18: error: request for member ‘push_back’ in ‘trajectory’, which is of non-class type ‘int’
       trajectory.push_back(pose);
                  ^
/home/juliocesar/catkin_ws/src/bezier-indigo-devel/bezier_library/src/bezier_library.cpp:677:16: error: request for member ‘push_back’ in ‘trajectory’, which is of non-class type ‘int’
     trajectory.push_back(pose);
                ^
/home/juliocesar/catkin_ws/src/bezier-indigo-devel/bezier_library/src/bezier_library.cpp: At global scope:
/home/juliocesar/catkin_ws/src/bezier-indigo-devel/bezier_library/src/bezier_library.cpp:683:43: error: variable or field ‘invertXAxisOfPoses’ declared void
 void Bezier::invertXAxisOfPoses(EigenSTL::vector_Affine3d &line)
                                           ^
/home/juliocesar/catkin_ws/src/bezier-indigo-devel/bezier_library/src/bezier_library.cpp:683:33: error: ‘vector_Affine3d’ is not a member of ‘EigenSTL’
 void Bezier::invertXAxisOfPoses(EigenSTL::vector_Affine3d &line)
                                 ^
/home/juliocesar/catkin_ws/src/bezier-indigo-devel/bezier_library/src/bezier_library.cpp:683:60: error: ‘line’ was not declared in this scope
 void Bezier::invertXAxisOfPoses(EigenSTL::vector_Affine3d &line)
                                                            ^
make[2]: *** [bezier-indigo-devel/bezier_library/CMakeFiles/bezier_library_obj.dir/src/bezier_library.cpp.o] Error 1
make[1]: *** [bezier-indigo-devel/bezier_library/CMakeFiles/bezier_library_obj.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
```

Additionally, `ros-indigo-eigen-stl-containers` dependency is already installed.

Thank you.

VictorLamoine, 2016-09-06:
Did you manually clone bezier in a new workspace?
Can you follow these instructions: https://github.com/ros-industrial-consortium/bezier#quick-help

jcgarciaca, 2016-09-06:
I manually cloned bezier in my workspace. According with the instructions, should I create a new workspace?

VictorLamoine, 2016-09-06:
Yes and follow the instructions on the main read-me page, if you have errors, please copy/paste them here!

jcgarciaca, 2016-09-08:
It worked, thank you.

closed 2016-09-08 by jcgarciaca