ros-industrial-attic/bezier #157: Process has died!

oriolorra, 2017-02-23:
Hi all,

I have a problem to execute this package. I have Ubuntu 14.04, and ROS-Indigo, of course. I downloaded this package and what it is needed to compile it. When I do catkin_make, everything is fine.

But when I execute this command : 
_"roslaunch bezier_application fanuc_m20ia_surfacing.launch surfacing_mode:=true mesh_cad:=plane/plane_defect.ply"_

I get this error (surrounded by =======):

* [ INFO] [1487859577.249922824]: Added FollowJointTrajectory controller for 
* [ INFO] [1487859577.250122977]: Returned 1 controllers in list
* [ INFO] [1487859577.294834033]: Trajectory execution is managing controllers
* Loading 'move_group/ApplyPlanningSceneService'...
* Loading 'move_group/ClearOctomapService'...
* Loading 'move_group/MoveGroupCartesianPathService'...
* Loading 'move_group/MoveGroupExecuteService'...
* Loading 'move_group/MoveGroupExecuteTrajectoryAction'...
* Loading 'move_group/MoveGroupGetPlanningSceneService'...
* Loading 'move_group/MoveGroupKinematicsService'...
* Loading 'move_group/MoveGroupMoveAction'...
* *================================================================================REQUIRED process [surfacing_application-5] has died!
* process has died [pid 18263, exit code -11, cmd     */home/oorra/catkin_ws/devel/lib/bezier_application/surfacing_application * __name:=surfacing_application __log:=/home/oorra/.ros/log/ce1a1068-f9c6-11e6-8931-* d0577b13869d/surfacing_application-5.log].
* log file: /home/oorra/.ros/log/ce1a1068-f9c6-11e6-8931-d0577b13869d/surfacing_application-5*.log
* Initiating shutdown!
* ===============================================================================* =
* Loading 'move_group/MoveGroupPickPlaceAction'...
* Loading 'move_group/MoveGroupPlanService'...
* Loading 'move_group/MoveGroupQueryPlannersService'...
* Loading 'move_group/MoveGroupStateValidationService'...
* [ INFO] [1487859577.565111774]: 

********************************************************
* MoveGroup using: 
*     - ApplyPlanningSceneService
*     - ClearOctomapService
*     - CartesianPathService
*     - ExecuteTrajectoryService
*     - ExecuteTrajectoryAction
*     - GetPlanningSceneService
*     - KinematicsService
*     - MoveAction
*     - PickPlaceAction
*     - MotionPlanService
*     - QueryPlannersService
*     - StateValidationService
********************************************************

[ INFO] [1487859577.565216638]: MoveGroup context using planning plugin ompl_interface/OMPLPlanner
[ INFO] [1487859577.565251098]: MoveGroup context initialization complete

Could anyone say me how to solve it? 

Thanks in advence,

Oriol

oriolorra, 2017-02-23:
I've been investigating, and I think that fails this instruction "if(!appendInputMesh(input_mesh))"  of bezier_grinding_surfacing.cpp at line 19 on bezier_library project.

Could it be the error ?

VictorLamoine, 2017-02-27:
I'll test that today!

VictorLamoine, 2017-02-27:
Works on my machine expect for this issue: https://github.com/ros-industrial-consortium/bezier/issues/78 (tested on Ubuntu 14.04 with ROS Indigo)
Can you provide a `gdb` trace?

- What VTK version are you using?
- What PCL version are you using?

[image]

oriolorra, 2017-02-28:
Thanks for your attention!! Today, I have been testing it again and I've found the problem.

I'm using Ubuntu 14.04 with ROS Indigo too, But, the rviz_visualization_tools (https://github.com/davetcoleman/rviz_visual_tools) must be on **kinetic-devel** an not **Indigo-devel**.

When I put **Indigo-devel** this line _reader = vtkSmartPointer<vtkPLYReader>::New();_ (line 27) on bezier_library.cpp fails. But with **kinetic-devel** all works well.

Best regards,

mlddg, 2017-06-07:
I have similar problem, but I use ROS Kinetic and the rviz_visualization_tools is on kinetic-devel.

this is the ERROR:

[ INFO] [1496827833.983787638]: MoveGroup context initialization complete

You can start planning now!

**[ INFO] [1496827834.527854819]: Topic '/rviz_visual_tools' waiting for subscriber...=REQUIRED process [surfacing_application-6] has died!
process has died [pid 15927, exit code -11, cmd /home/lmy/bezier_ws/devel/lib/bezier_application/surfacing_application __name:=surfacing_application __log:=/home/lmy/.ros/log/f3646a8c-4b63-11e7-9993-484520ac8e0b/surfacing_application-6.log].
log file: /home/lmy/.ros/log/f3646a8c-4b63-11e7-9993-484520ac8e0b/surfacing_application-6*.log
Initiating shutdown!**

[rviz-7] killing on exit
[surfacing_application-6] killing on exit
[move_group-5] killing on exit
[robot_state_publisher-4] killing on exit
[joint_trajectory_action-3] killing on exit
[industrial_robot_simulator-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

can you help me? Actually I am a freshman about ROS.
Thanks very much

VictorLamoine, 2017-06-07:
Closing as this is old, @mlddg i'll answer to you in your new issue: #158

closed 2017-06-07 by VictorLamoine