ros-industrial-attic/bezier #158: REQUIRED process [surfacing_application-6] has died!

mlddg, 2017-06-07:
Hello everyone:
I have problem when execute:

 roslaunch bezier_application fanuc_m20ia_surfacing.launch surfacing_mode:=true mesh_cad:=plane/plane_defect.ply

I tested it on **Ubuntu 16.04, ROS-kinetic, PCL 1.8.0, VTK 7.0.0.**

This is the ERROR:

... logging to /home/lmy/.ros/log/f3646a8c-4b63-11e7-9993-484520ac8e0b/roslaunch-lmy-PC-15857.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://lmy-PC:46069/

SUMMARY
========

PARAMETERS
 * /controller_joint_names: ['joint_1', 'join...
 * /mesh_cad_param: plane/plane_defec...
 * /mesh_defect_param: 
 * /move_group/allow_trajectory_execution: True
 * /move_group/capabilities: move_group/MoveGr...
 * /move_group/controller_list: [{'action_ns': 'j...
 * /move_group/jiggle_fraction: 0.05
 * /move_group/manipulator/default_planner_config: RRTConnectkConfig...
 * /move_group/manipulator/longest_valid_segment_fraction: 0.05
 * /move_group/manipulator/planner_configs: ['SBLkConfigDefau...
 * /move_group/manipulator/projection_evaluator: joints(joint_1,jo...
 * /move_group/max_range: 5.0
 * /move_group/max_safe_path_cost: 1
 * /move_group/moveit_controller_manager: moveit_simple_con...
 * /move_group/moveit_manage_controllers: True
 * /move_group/octomap_resolution: 0.025
 * /move_group/planner_configs/BKPIECEkConfigDefault/border_fraction: 0.9
 * /move_group/planner_configs/BKPIECEkConfigDefault/failed_expansion_score_factor: 0.5
 * /move_group/planner_configs/BKPIECEkConfigDefault/min_valid_path_fraction: 0.5
 * /move_group/planner_configs/BKPIECEkConfigDefault/range: 0.0
 * /move_group/planner_configs/BKPIECEkConfigDefault/type: geometric::BKPIECE
 * /move_group/planner_configs/ESTkConfigDefault/goal_bias: 0.05
 * /move_group/planner_configs/ESTkConfigDefault/range: 0.0
 * /move_group/planner_configs/ESTkConfigDefault/type: geometric::EST
 * /move_group/planner_configs/KPIECEkConfigDefault/border_fraction: 0.9
 * /move_group/planner_configs/KPIECEkConfigDefault/failed_expansion_score_factor: 0.5
 * /move_group/planner_configs/KPIECEkConfigDefault/goal_bias: 0.05
 * /move_group/planner_configs/KPIECEkConfigDefault/min_valid_path_fraction: 0.5
 * /move_group/planner_configs/KPIECEkConfigDefault/range: 0.0
 * /move_group/planner_configs/KPIECEkConfigDefault/type: geometric::KPIECE
 * /move_group/planner_configs/LBKPIECEkConfigDefault/border_fraction: 0.9
 * /move_group/planner_configs/LBKPIECEkConfigDefault/min_valid_path_fraction: 0.5
 * /move_group/planner_configs/LBKPIECEkConfigDefault/range: 0.0
 * /move_group/planner_configs/LBKPIECEkConfigDefault/type: geometric::LBKPIECE
 * /move_group/planner_configs/PRMkConfigDefault/max_nearest_neighbors: 10
 * /move_group/planner_configs/PRMkConfigDefault/type: geometric::PRM
 * /move_group/planner_configs/PRMstarkConfigDefault/type: geometric::PRMstar
 * /move_group/planner_configs/RRTConnectkConfigDefault/range: 0.0
 * /move_group/planner_configs/RRTConnectkConfigDefault/type: geometric::RRTCon...
 * /move_group/planner_configs/RRTkConfigDefault/goal_bias: 0.05
 * /move_group/planner_configs/RRTkConfigDefault/range: 0.0
 * /move_group/planner_configs/RRTkConfigDefault/type: geometric::RRT
 * /move_group/planner_configs/RRTstarkConfigDefault/delay_collision_checking: 1
 * /move_group/planner_configs/RRTstarkConfigDefault/goal_bias: 0.05
 * /move_group/planner_configs/RRTstarkConfigDefault/range: 0.0
 * /move_group/planner_configs/RRTstarkConfigDefault/type: geometric::RRTstar
 * /move_group/planner_configs/SBLkConfigDefault/range: 0.0
 * /move_group/planner_configs/SBLkConfigDefault/type: geometric::SBL
 * /move_group/planner_configs/TRRTkConfigDefault/frountierNodeRatio: 0.1
 * /move_group/planner_configs/TRRTkConfigDefault/frountier_threshold: 0.0
[... 140 lines cut ...]
[ INFO] [1496827833.983787638]: MoveGroup context initialization complete

You can start planning now!

[ INFO] [1496827834.527854819]: Topic '/rviz_visual_tools' waiting for subscriber...
================================================================================REQUIRED process [surfacing_application-6] has died!
process has died [pid 15927, exit code -11, cmd /home/lmy/bezier_ws/devel/lib/bezier_application/surfacing_application __name:=surfacing_application __log:=/home/lmy/.ros/log/f3646a8c-4b63-11e7-9993-484520ac8e0b/surfacing_application-6.log].
log file: /home/lmy/.ros/log/f3646a8c-4b63-11e7-9993-484520ac8e0b/surfacing_application-6*.log
Initiating shutdown!
================================================================================
[rviz-7] killing on exit
[surfacing_application-6] killing on exit
[move_group-5] killing on exit
[robot_state_publisher-4] killing on exit
[joint_trajectory_action-3] killing on exit
[industrial_robot_simulator-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

Can anyone help me?
Thanks very much.
mlddg

VictorLamoine, 2017-06-07:
Related: #157

VictorLamoine, 2017-06-07:
I have tested on Ubuntu 16.04 and ROS Kinetic and it works for me.

Can you debug the node with GDB please?
http://wiki.ros.org/roslaunch/Tutorials/Roslaunch%20Nodes%20in%20Valgrind%20or%20GDB

Don't forget to build the catkin workspace in debug mode!

```bash
catkin_make -DCMAKE_BUILD_TYPE=Debug
```

Please post the `gdb` output here

mlddg, 2017-06-07:
@VictorLamoine Thank you for your help
I have built the catkin workspace by **catkin bulid**, followed as Quick help in https://github.com/ros-industrial-consortium/bezier

Did you mean that I should build it by `catkin_make -DCMAKE_BUILD_TYPE=Debug` again ？

VictorLamoine, 2017-06-07:
Yes, build in `Debug` mode rather than `Release`, this will enable you to debug with `gdb`

mlddg, 2017-06-07:
@VictorLamoine But when I use catkin_make , the following error occured.
```
/home/lmy/bezier_ws/src/bezier/bezier_library/src/bezier_library.cpp:1087:22:   required from here
/usr/include/vtk-6.2/vtkSmartPointer.h:94:40: error: invalid static_cast from type ‘vtkObjectBase* const’ to type ‘vtkCell*’
     return static_cast<T*>(this->Object);
                                        ^
/usr/include/vtk-6.2/vtkSmartPointer.h: In instantiation of ‘static vtkSmartPointer<T> vtkSmartPointer<T>::New() [with T = vtkIdList]’:
/home/lmy/bezier_ws/src/bezier/bezier_library/src/bezier_library.cpp:1094:80:   required from here
/usr/include/vtk-6.2/vtkSmartPointer.h:117:37: error: incomplete type ‘vtkIdList’ used in nested name specifier
     return vtkSmartPointer<T>(T::New(), NoReference());
                                     ^
/usr/include/vtk-6.2/vtkSmartPointer.h: In instantiation of ‘vtkSmartPointer<T>::operator T*() const [with T = vtkIdList]’:
/home/lmy/bezier_ws/src/bezier/bezier_library/src/bezier_library.cpp:1095:81:   required from here
/usr/include/vtk-6.2/vtkSmartPointer.h:79:40: error: invalid static_cast from type ‘vtkObjectBase* const’ to type ‘vtkIdList*’
     return static_cast<T*>(this->Object);
                                        ^
/usr/include/vtk-6.2/vtkSmartPointer.h: In instantiation of ‘T* vtkSmartPointer<T>::operator->() const [with T = vtkIdList]’:
/home/lmy/bezier_ws/src/bezier/bezier_library/src/bezier_library.cpp:1106:48:   required from here
/usr/include/vtk-6.2/vtkSmartPointer.h:94:40: error: invalid static_cast from type ‘vtkObjectBase* const’ to type ‘vtkIdList*’
     return static_cast<T*>(this->Object);
                                        ^
[ 70%] Building CXX object rviz_visual_tools/CMakeFiles/rviz_visual_tools_gui.dir/src/rviz_visual_tools_gui.cpp.o
bezier/bezier_library/CMakeFiles/bezier_library_obj.dir/build.make:62: recipe for target 'bezier/bezier_library/CMakeFiles/bezier_library_obj.dir/src/bezier_library.cpp.o' failed
make[2]: *** [bezier/bezier_library/CMakeFiles/bezier_library_obj.dir/src/bezier_library.cpp.o] Error 1
CMakeFiles/Makefile2:9776: recipe for target 'bezier/bezier_library/CMakeFiles/bezier_library_obj.dir/all' failed
make[1]: *** [bezier/bezier_library/CMakeFiles/bezier_library_obj.dir/all] Error 2
```

VictorLamoine, 2017-06-07:
There's a problem because it's including VTK 6.2 instead of VTK 7!
I don't have time this afternoon but I'll help you tomorrow morning.

mlddg, 2017-06-07:
@VictorLamoine Thank you very much for your help!
I make the VTK7.0 again and it works.
And I debug the node with GBD. This shows:

```
... logging to /home/lmy/.ros/log/295c4eee-4b95-11e7-a9fa-484520ac8e0b/roslaunch-lmy-PC-4934.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://lmy-PC:39325/

SUMMARY
========

PARAMETERS
 * /controller_joint_names: ['joint_1', 'join...
 * /mesh_cad_param: plane/plane_defec...
 * /mesh_defect_param: 
 * /move_group/allow_trajectory_execution: True
 * /move_group/capabilities: move_group/MoveGr...
 * /move_group/controller_list: [{'action_ns': 'j...
 * /move_group/jiggle_fraction: 0.05
 * /move_group/manipulator/default_planner_config: RRTConnectkConfig...
 * /move_group/manipulator/longest_valid_segment_fraction: 0.05
 * /move_group/manipulator/planner_configs: ['SBLkConfigDefau...
 * /move_group/manipulator/projection_evaluator: joints(joint_1,jo...
 * /move_group/max_range: 5.0
 * /move_group/max_safe_path_cost: 1
 * /move_group/moveit_controller_manager: moveit_simple_con...
 * /move_group/moveit_manage_controllers: True
 * /move_group/octomap_resolution: 0.025
 * /move_group/planner_configs/BKPIECEkConfigDefault/border_fraction: 0.9
 * /move_group/planner_configs/BKPIECEkConfigDefault/failed_expansion_score_factor: 0.5
 * /move_group/planner_configs/BKPIECEkConfigDefault/min_valid_path_fraction: 0.5
 * /move_group/planner_configs/BKPIECEkConfigDefault/range: 0.0
 * /move_group/planner_configs/BKPIECEkConfigDefault/type: geometric::BKPIECE
 * /move_group/planner_configs/ESTkConfigDefault/goal_bias: 0.05
 * /move_group/planner_configs/ESTkConfigDefault/range: 0.0
 * /move_group/planner_configs/ESTkConfigDefault/type: geometric::EST
 * /move_group/planner_configs/KPIECEkConfigDefault/border_fraction: 0.9
 * /move_group/planner_configs/KPIECEkConfigDefault/failed_expansion_score_factor: 0.5
 * /move_group/planner_configs/KPIECEkConfigDefault/goal_bias: 0.05
 * /move_group/planner_configs/KPIECEkConfigDefault/min_valid_path_fraction: 0.5
 * /move_group/planner_configs/KPIECEkConfigDefault/range: 0.0
 * /move_group/planner_configs/KPIECEkConfigDefault/type: geometric::KPIECE
 * /move_group/planner_configs/LBKPIECEkConfigDefault/border_fraction: 0.9
 * /move_group/planner_configs/LBKPIECEkConfigDefault/min_valid_path_fraction: 0.5
 * /move_group/planner_configs/LBKPIECEkConfigDefault/range: 0.0
 * /move_group/planner_configs/LBKPIECEkConfigDefault/type: geometric::LBKPIECE
 * /move_group/planner_configs/PRMkConfigDefault/max_nearest_neighbors: 10
 * /move_group/planner_configs/PRMkConfigDefault/type: geometric::PRM
 * /move_group/planner_configs/PRMstarkConfigDefault/type: geometric::PRMstar
 * /move_group/planner_configs/RRTConnectkConfigDefault/range: 0.0
 * /move_group/planner_configs/RRTConnectkConfigDefault/type: geometric::RRTCon...
 * /move_group/planner_configs/RRTkConfigDefault/goal_bias: 0.05
 * /move_group/planner_configs/RRTkConfigDefault/range: 0.0
 * /move_group/planner_configs/RRTkConfigDefault/type: geometric::RRT
 * /move_group/planner_configs/RRTstarkConfigDefault/delay_collision_checking: 1
 * /move_group/planner_configs/RRTstarkConfigDefault/goal_bias: 0.05
 * /move_group/planner_configs/RRTstarkConfigDefault/range: 0.0
 * /move_group/planner_configs/RRTstarkConfigDefault/type: geometric::RRTstar
 * /move_group/planner_configs/SBLkConfigDefault/range: 0.0
 * /move_group/planner_configs/SBLkConfigDefault/type: geometric::SBL
 * /move_group/planner_configs/TRRTkConfigDefault/frountierNodeRatio: 0.1
 * /move_group/planner_configs/TRRTkConfigDefault/frountier_threshold: 0.0
 * /move_group/planner_configs/TRRTkConfigDefault/goal_bias: 0.05
 * /move_group/planner_configs/TRRTkConfigDefault/init_temperature: 10e-6
 * /move_group/planner_configs/TRRTkConfigDefault/k_constant: 0.0
 * /move_group/planner_configs/TRRTkConfigDefault/max_states_failed: 10
[... 150 lines cut ...]

[ INFO] [1496848969.934493119]: MoveGroup context using planning plugin ompl_interface/OMPLPlanner
[ INFO] [1496848969.934521123]: MoveGroup context initialization complete

You can start planning now!

[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[tcsetpgrp failed in terminal_inferior: 对设备不适当的 ioctl 操作]
[tcsetpgrp failed in terminal_inferior: 对设备不适当的 ioctl 操作]
[New Thread 0x7fffbcbe0700 (LWP 5613)]
[New Thread 0x7fffb7fff700 (LWP 5614)]
[New Thread 0x7fffb77fe700 (LWP 5615)]
[New Thread 0x7fffb6ffd700 (LWP 5620)]
[tcsetpgrp failed in terminal_inferior: 对设备不适当的 ioctl 操作]
[tcsetpgrp failed in terminal_inferior: 对设备不适当的 ioctl 操作]
[ INFO] [1496848975.139416236]: CAD mesh file:plane/plane_defect.ply
[ INFO] [1496848975.139550737]: Bezier::Bezier: RViz visualization tool is initialized in 'base' and the topic name is 'rviz_visual_tools'
[ WARN] [1496848975.142364083]: Waiting for a subscriber on topic "/bezier_cad_mesh"
[ INFO] [1496848976.145417373]: Topic '/rviz_visual_tools' waiting for subscriber...

Thread 1 "surfacing_appli" received signal SIGSEGV, Segmentation fault.
0x00007fffee091be7 in pcl::VTKUtils::vtk2mesh(vtkSmartPointer<vtkPolyData> const&, pcl::PolygonMesh&) () from /usr/local/lib/libpcl_surface.so.1.8
(gdb) 
```

VictorLamoine, 2017-06-07:
The crash happens in PCL and is VTK related
We need to make sure you have compiled PCL 1.8 against VTK 7.0

To verify that: (if you don't have `ccmake`: `sudo apt install -y cmake-curses-gui`)
```
cd pcl/build_release/
ccmake .
```

Search for this line:
```
VTK_DIR           /usr/local/lib/cmake/vtk-7.0
```

It must point to the VTK 7.0 you compiled, not any other version.
If it's not right, change it to the right path.

Configure (`c` key) / generate (`g`) again the CMake configuration (see the bottom instructions in `ccmake`).

After that compile/install PCL again.
When done, compile your catkin workspace and try again the `roslaunch` command.

mlddg, 2017-06-07:
And I have another question about the bezier.rosinstall.
https://github.com/ros-industrial-consortium/bezier/blob/kinetic-devel/bezier.rosinstall
In this kinetic-devel, the version of bezier is indigo. The first time I use this **rosinstall** and it had problem about the **moveit/move_group_interface/move_group.h**
So I manually change the indigo to kinetic and it worked.

mlddg, 2017-06-08:
@VictorLamoine Thank you very much! All of the things work well now!

VictorLamoine, 2017-06-08:
About the move_group include, there has been a change between indigo and kinetic, the change you made is OK.

Thanks for notifying me it works :beer:

closed 2017-06-08 by VictorLamoine