ros-industrial-attic/bezier #159: Could not generate trajectory: Failed to extract upper part of dilated mesh

lmyhit, 2017-06-14:
Hello everyone,
When I use my own ply files as the mash file, the following error occured.
This is my files. **2.ply** is the real file generated from the 3-D scanner,**3.STL** is a test file generated by Solidworks. Both of them had the same error.
[a.zip](https://github.com/ros-industrial-consortium/bezier/files/1073842/a.zip)

the ERROR is : 
```
You can start planning now!

[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[tcsetpgrp failed in terminal_inferior: 对设备不适当的 ioctl 操作]
[tcsetpgrp failed in terminal_inferior: 对设备不适当的 ioctl 操作]
[New Thread 0x7fffc69cd700 (LWP 6939)]
[New Thread 0x7fffc61cc700 (LWP 6940)]
[New Thread 0x7fffc59cb700 (LWP 6941)]
[New Thread 0x7fffc51ca700 (LWP 6946)]
[tcsetpgrp failed in terminal_inferior: 对设备不适当的 ioctl 操作]
[tcsetpgrp failed in terminal_inferior: 对设备不适当的 ioctl 操作]
[ INFO] [1497425553.612233982]: CAD mesh file:a/3.STL
[ INFO] [1497425553.613135515]: Bezier::Bezier: RViz visualization tool is initialized in 'base' and the topic name is 'rviz_visual_tools'
[ WARN] [1497425553.621250754]: Waiting for a subscriber on topic "/bezier_cad_mesh"
[ INFO] [1497425554.625100236]: Topic '/rviz_visual_tools' waiting for subscriber...
[ INFO] [1497425554.843491929]: BezierGrindingSurfacing::generateTrajectory: Dilating mesh
[ INFO] [1497425554.844126909]: Bezier::dilate: default parameters will be used to dilate the mesh
[New Thread 0x7fffb7fff700 (LWP 6953)]
[New Thread 0x7fffb77fe700 (LWP 6954)]
[New Thread 0x7fffb6ffd700 (LWP 6955)]
[Thread 0x7fffb6ffd700 (LWP 6955) exited]
[Thread 0x7fffb77fe700 (LWP 6954) exited]
[Thread 0x7fffb7fff700 (LWP 6953) exited]
[ WARN] [1497425554.948101954]: Bezier::keepUpperPartofDilatedMesh: Computing normals...
[ WARN] [1497425554.959642432]: Empty trajectory!
[ERROR] [1497425554.959817289]: Could not generate trajectory:
Failed to extract upper part of dilated mesh

```
And I have another 2 questions, 
1. How can I generate the proper mesh file? I generated it from Geomagic software and Solidworks.
2. How can I locate the mesh in the space, to edit it in the mesh file or can I make a transformation in the Bezier codes?
Thanks very much!
@VictorLamoine

VictorLamoine, 2017-06-15:
1. Your mesh looks good.
2. You can move the mesh with blender or with meshlab: 
- https://wiki.blender.org/index.php/Doc:2.4/Manual/3D_interaction/Transformations/Basics/Grab
- https://www.youtube.com/watch?v=gJKv-nC6GbY

The problem is the following: `Failed to extract upper part of dilated mesh`
Here is a quick explanation of what it does:
https://github.com/ros-industrial-consortium/bezier/blob/kinetic-devel/bezier_library/doc/README_keep_upper_part_of_dilated_mesh.md

The method code is here:
https://github.com/ros-industrial-consortium/bezier/blob/kinetic-devel/bezier_library/src/bezier_library.cpp#L1053

You need to determine why this function fails.
I suggest step by step debugging with gdb.

lmyhit, 2017-06-20:
@VictorLamoine 
This problem has been solved. It was caused by the unit settings of the mesh files, the mesh I set is mm and the right unit is m.
Thanks for your help!

VictorLamoine, 2017-06-20:
Thanks for the feedback, I'm not sure I'll have time to answer to your other question today! I'll look into it a little bit later

closed 2017-06-20 by VictorLamoine