ros-industrial-attic/bezier #161: Required Process has died!

andreagen, 2017-09-29:
Hi everyone,

I have a problem to run the package.
I downloaded this package and what it is needed to compile it. When I do catkin build, I have no problem, everything is fine.
I have Ubuntu 14.04, ROS Indigo.

But when I execute this command :
"roslaunch bezier_application fanuc_m20ia_surfacing.launch surfacing_mode:=true mesh_cad:=plane/plane_defect.ply"

The RViz window open and then closed when it is computing normals with this error:

[ INFO] [1506681431.925214949]: Topic '/rviz_visual_tools' waiting for subscriber...
[ INFO] [1506681432.137481747]: BezierGrindingSurfacing::generateTrajectory: Dilating mesh
[ INFO] [1506681432.137811386]: Bezier::dilate: default parameters will be used to dilate the mesh
[ WARN] [1506681433.532489843]: Bezier::keepUpperPartofDilatedMesh: Computing normals...
[ INFO] [1506681441.983707720]: BezierGrindingSurfacing::generateTrajectory: Generating grinding trajectories
[ INFO] [1506681442.026175666]: BezierGrindingSurfacing::generateTrajectory: Generating extrication trajectories
[ERROR] [1506681442.114268992]: BezierGrindingSurfacing::generateTrajectory: Extrication stripper has 2 lines
[ INFO] [1506681442.431377219]: Trajectory size: 1507
[ INFO] [1506681442.441064979]: Loading robot model 'fanuc_m20ia'...
surfacing_application: /usr/include/eigen3/Eigen/src/SVD/JacobiSVD.h:632: const MatrixVType& Eigen::JacobiSVD<MatrixType, QRPreconditioner>::matrixV() const [with _MatrixType = Eigen::Matrix<double, 3, 3>; int QRPreconditioner = 2; Eigen::JacobiSVD<MatrixType, QRPreconditioner>::MatrixVType = Eigen::Matrix<double, 3, 3>; typename _MatrixType::Scalar = double]: Assertion `computeV() && "This JacobiSVD decomposition didn't compute V. Did you ask for it?"' failed.
================================================================================REQUIRED process [surfacing_application-6] has died!
process has died [pid 11044, exit code -6, cmd /home/agenzel/catkin_ws/devel/lib/bezier_application/surfacing_application __name:=surfacing_application __log:=/home/agenzel/.ros/log/2562688e-a502-11e7-80e8-180373adfdd9/surfacing_application-6.log].
log file: /home/agenzel/.ros/log/2562688e-a502-11e7-80e8-180373adfdd9/surfacing_application-6*.log
Initiating shutdown!
================================================================================
[rviz-7] killing on exit
[surfacing_application-6] killing on exit
[move_group-5] killing on exit
[robot_state_publisher-4] killing on exit
[joint_trajectory_action-3] killing on exit
[industrial_robot_simulator-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete

I have installed Visualization Toolkit version 7.1.1 and Point Cloud Library version 1.8.1
And I have this version of libeigen3 :

$ dpkg -l libeigen3-dev
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name           Version      Architecture Description
+++-==============-============-============-=================================
ii  libeigen3-dev  3.2.5-4~ubun all          lightweight C++ template library 

It's almost the same problem as here: [https://github.com/ros-industrial-consortium/bezier/issues/78](url)

Could anyone help me to solve it ?
Thanks!

Andrea

VictorLamoine, 2017-10-06:
It's the same problem we had before, I have never got an explanation for this.
Can you try installing an other Eigen version ?

VictorLamoine, 2017-10-06:
Closing in favour of #78 
Please continue the discussion on the issue above!

closed 2017-10-06 by VictorLamoine