ros-industrial-attic/bezier #164: data is not aligned"' failed.

tstiger, 2018-04-08:
[ INFO] [1523175971.147508874]: MoveGroup context using planning plugin ompl_interface/OMPLPlanner
[ INFO] [1523175971.147707567]: MoveGroup context initialization complete

You can start planning now!

[ INFO] [1523175971.408097734]: CAD mesh file:plane/plane_defect.ply
[ INFO] [1523175971.411535981]: Bezier::Bezier: RViz visualization tool is initialized in 'base' and the topic name is 'rviz_visual_tools'
[ WARN] [1523175971.529051466]: Waiting for a subscriber on topic "/bezier_cad_mesh"
[ WARN] [1523175972.529266322]: Waiting for a subscriber on topic "/bezier_cad_mesh"
[ WARN] [1523175973.529467687]: Waiting for a subscriber on topic "/bezier_cad_mesh"
[ INFO] [1523175974.531350606]: Topic '/rviz_visual_tools' waiting for subscriber...
[ INFO] [1523175974.927955538]: BezierGrindingSurfacing::generateTrajectory: Dilating mesh
[ INFO] [1523175974.930134351]: Bezier::dilate: default parameters will be used to dilate the mesh
[ WARN] [1523175976.720460113]: Bezier::keepUpperPartofDilatedMesh: Computing normals...
[ INFO] [1523175980.525965529]: BezierGrindingSurfacing::generateTrajectory: Generating grinding trajectories
bezier_application_surfacing: /usr/include/eigen3/Eigen/src/Core/CoreEvaluators.h:838: Eigen::internal::block_evaluator<ArgType, BlockRows, BlockCols, InnerPanel, true>::block_evaluator(const XprType&) [with ArgType = Eigen::Matrix<double, 4, 4>; int BlockRows = 3; int BlockCols = 1; bool InnerPanel = true; Eigen::internal::block_evaluator<ArgType, BlockRows, BlockCols, InnerPanel, true>::XprType = Eigen::Block<Eigen::Matrix<double, 4, 4>, 3, 1, true>]: Assertion `((size_t(block.data()) % (((int)1 >= (int)evaluator<XprType>::Alignment) ? (int)1 : (int)evaluator<XprType>::Alignment)) == 0) && "data is not aligned"' failed.
================================================================================REQUIRED process [bezier_application_surfacing-6] has died!
process has died [pid 9514, exit code -6, cmd /home/ros-industrial/catkin_qtws/devel/lib/bezier_application/bezier_application_surfacing __name:=bezier_application_surfacing __log:=/home/ros-industrial/.ros/log/7bb49618-3b06-11e8-a648-080027883dcb/bezier_application_surfacing-6.log].
log file: /home/ros-industrial/.ros/log/7bb49618-3b06-11e8-a648-080027883dcb/bezier_application_surfacing-6*.log
Initiating shutdown!
================================================================================
[rviz-7] killing on exit
[bezier_application_surfacing-6] killing on exit
[move_group-5] killing on exit
[robot_state_publisher-4] killing on exit
[joint_trajectory_action-3] killing on exit
[industrial_robot_simulator-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

tstiger, 2018-04-09:
Beizier_library.hpp
#define EIGEN_DONT_ALIGN

I added this macro, in beizier_library.hpp.
it's work, but  
in surfacing.cpp
line 150
   // Bezier trajectory
    if (group.computeCartesianPath(way_points_msg, 0.05, 0.0, srv.request.trajectory) < 0.95)
    {
      ROS_ERROR_STREAM(
          "Bezier application: A solution could not be found to move the robot along the trajectory, aborting.");
      break;
    }
case break , ros_error.

tstiger, 2018-04-09:
dpkg -l |grep libeigen*
ii  libeigen3-dev                                3.3~beta1-2                                           all          lightweight C++ template library for linear algebra

VictorLamoine, 2019-02-14:
- I made sure that all Eigen structures are passed by reference
- I checked that every vector makes sure the alignment is correct

I'm not sure if `std::pair` with Eigen structures requires a specific alignment:

```bash
$ grep -rn "std::pair<Eigen"
include/bezier/bezier_library.hpp:79:  typedef std::pair<Eigen::Vector3d, Eigen::Vector3d> PointNormal;
include/bezier/params_observer.hpp:117:  std::pair<Eigen::Vector3d, Eigen::Vector3d> slicing_orientation_;
```

This might be the cause of the crash