ros-industrial-attic/bezier #166: Process[surfacing_application-5] has died!

GaoKui, 2018-12-21:
HELLO:
I tested it on Ubuntu 16.04, ROS-kinetic, PCL 1.8.0, VTK 7.1.1.
When I use GDB, it  shows:

[WARN] [1545377664.210452821]: Bezier::keepUpperPartofDilatedMesh: Computing normals...
[ INFO] [1545377666.796006527]: BezierGrindingSurfacing::generateTrajectory: Generating grinding trajectories
[ INFO] [1545377666.817293583]: BezierGrindingSurfacing::generateTrajectory: Generating extrication trajectories
[ERROR] [1545377666.836747402]: BezierGrindingSurfacing::generateTrajectory: Extrication stripper has 2 lines

Thread 1 "bezier_applicat" received signal SIGSEGV, Segmentation fault.
0x000000000042a5c2 in _mm256_load_pd (__P=0x7fffffffbb10)
    at /usr/lib/gcc/x86_64-linux-gnu/5/include/avxintrin.h:836
836	  return *(__m256d *)__P;
(gdb)

GaoKui, 2018-12-21:
Thank you very much! @VictorLamoine

VictorLamoine, 2018-12-21:
The information you have provided is not enough.

Please provide a full backtrace, for that you need to compile in debug mode and produce a backtrace after the crash.

GaoKui, 2018-12-21:
#5  0x0000000000432c9d in Eigen::internal::generic_dense_assignment_kernel<Eigen::internal::evaluator<Eigen::Matrix<double, 4, 4, 0, 4, 4> >, Eigen::internal::evaluator<Eigen::Matrix<double, 4, 4, 0, 4, 4> >, Eigen::internal::assign_op<double>, 0>::assignPacketByOuterInner<16, 32, double __vector(4)>(long, long) (
---Type <return> to continue, or q <return> to quit---
    this=0x7fffffffba30, outer=0, inner=0)
    at /usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:606
#6  0x0000000000432a68 in Eigen::internal::copy_using_evaluator_innervec_CompleteUnrolling<Eigen::internal::generic_dense_assignment_kernel<Eigen::internal::evaluator<Eigen::Matrix<double, 4, 4, 0, 4, 4> >, Eigen::internal::evaluator<Eigen::Matrix<double, 4, 4, 0, 4, 4> >, Eigen::internal::assign_op<double>, 0>, 0, 16>::run (kernel=...) at /usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:244
#7  0x00007ffff50bf686 in rviz_visual_tools::RvizVisualTools::publishCylinder(Eigen::Matrix<double, 3, 1, 0, 3, 1> const&, Eigen::Matrix<double, 3, 1, 0, 3, 1> const&, std_msgs::ColorRGBA_<std::allocator<void> > const&, double, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) ()
   from /opt/ros/kinetic/lib/librviz_visual_tools.so
#8  0x00007ffff50bfa1f in rviz_visual_tools::RvizVisualTools::publishCylinder(Eigen::Matrix<double, 3, 1, 0, 3, 1> const&, Eigen::Matrix<double, 3, 1, 0, 3, 1> const&, rviz_visual_tools::colors, double, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) ()
   from /opt/ros/kinetic/lib/librviz_visual_tools.so
#9  0x00007ffff50c06ac in rviz_visual_tools::RvizVisualTools::publishPath(std::vector<Eigen::Matrix<double, 3, 1, 0, 3, 1>, Eigen::aligned_allocator<Eigen::Matrix<double, 3, 1, 0, 3, 1> > > const&, rviz_visual_tools::colors, double, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) () from /opt/ros/kinetic/lib/librviz_visual_tools.so

#10 0x00007ffff5d2a71f in Bezier::displayTrajectory (this=0x7242a0, 
---Type <return> to continue, or q <return> to quit---
    trajectory=std::vector of length 17, capacity 17 = {...}, 
    color=rviz_visual_tools::BLUE, display_normals=true, display_labels=false)
    at /home/walden/bezier_ws/src/bezier_library/src/bezier_library.cpp:151

#11 0x00007ffff7b9e281 in BezierGrindingSurfacing::generateTrajectory[abi:cxx11](std::vector<Eigen::Transform<double, 3, 2, 0>, Eigen::aligned_allocator<Eigen::Transform<double, 3, 2, 0> > >&, std::vector<bool, std::allocator<bool> >&, bool) (this=0x7242a0, trajectory=std::vector of length 0, capacity 0, 
    is_grinding_pose=std::vector<bool> of length 0, capacity 0, 
    display_markers=true)
    at /home/walden/bezier_ws/src/bezier_library/src/bezier_surfacing.cpp:441
#12 0x0000000000427c7f in main (argc=1, argv=0x7fffffffdb48)
    at /home/walden/bezier_ws/src/bezier_application/src/surfacing.cpp:105
@gavanderhoorn Thanks very much!

VictorLamoine, 2018-12-21:
So this is a case we do not handle in the library, when we extract lines from the mesh (by slicing it) we expect one and only one line to be produced.

This is not always the case (usually happens when the mesh has a hole).

You can find some comments here: https://github.com/ros-industrial-consortium/bezier/blob/kinetic-devel/bezier_library/src/bezier_surfacing.cpp#L154

I has merged a pull request that re-activates the commented error handling.

If you want to make this works you will have to handle this case, which means you have to dig into the code, understand it and add some functions to handle this case. The https://github.com/ros-industrial-consortium/bezier/tree/kinetic-devel/bezier_library/doc is a good place to start for understanding the algorithm.

closed 2018-12-21 by VictorLamoine

VictorLamoine, 2019-01-02:
I've just tested on Melodic and there is no crash. (I'm using the latest RViz visual tools version so https://github.com/ros-industrial-consortium/bezier/pull/168 is necessary is order to be able to compile the package).

[image]

I get an error because of the outdated frame names and robot packages but the trajectory is correctly generated.
```
[ERROR] [1546422903.365594211]: TF Problem: Invalid argument "/base_link" passed to lookupTransform argument target_frame in tf2 frame_ids cannot start with a '/' like: 
[ERROR] [1546422903.365624364]: Error encountered transforming waypoints to frame 'base_link'
[ERROR] [1546422903.365772260]: Bezier application: A solution could not be found to move the robot along the trajectory, aborting.

```

AnsunWu, 2020-09-29:
Hello Mr.Victor,  @VictorLamoine 

Hope you are doing well.
I'm stuck into this problem and hope to get your help, plz.

I'm trying to run the **'Bezier Example'** project through the launch file in the [link ](https://github.com/ros-industrial-consortium/bezier_examples):
 **_'fanuc_m10ia_surfacing.launch'_  and _''fanuc_m10ia_painting.launch ''_.**

**Since the Bezier project is very precious learning material for me **(almost the only one)**, I tried to use many different Ubuntu and ROS environments to test it.**

The **same** issue appeared in every environment that I tested. Detailed info is shown below:

> 
> **The environments that I tested** 
> **The project was compiled normally** in all these environments below:
> 
> **1. (On Vmware)**
>       Ubuntu 16.04 with ROS kinetic, QT 5.1.1, VTK8.2.0 (shown in the **'bezier repo'**), PCL 1.11.1(Latest version issue #12[as](https://github.com/ros-industrial-consortium/bezier/issues/12) )) ).
> 
> **2. (On Laptop hardware+ Nvidia GTX1050)**
>      Ubuntu 16.04 with ROS kinetic, QT 5.1.1, VTK8.2.0 (shown in the **'bezier repo'**), PCL 1.11.1(Latest version issue #12 [as](https://github.com/ros-industrial-consortium/bezier/issues/12) )) ).
> 
> **2. (On Vmware)**
>  Ubuntu 18.04 with ROS Melodic, QT 5.1.1, VTK8.2.0 (shown in the **'bezier repo'**), PCL 1.11.1(Latest version issue #12[as](https://github.com/ros-industrial-consortium/bezier/issues/12) )) ).
> 
> 

**Firstly**, the RVIZ interface showed up, the _robot model_ and the _mesh cad_ file showed up normally, like below:
[image]

And after for a while (**about 5 seconds**), the process was ended with an error like below:
[image]

During the normal period, I saw there were some warning that showed inside the RVIZ interface as below
(the yellow exclamation mark !):

> Uninitialized quaternion, assuming identity.
> scale.y and scale.z of LINE_LIST or LINE_STRIP are ignored.

**I have carefully checked all the repos below but still not be able to solve this issue:**
> [https://github.com/ros-industrial-consortium/bezier/issues/174](issue #174)
> [https://github.com/ros-industrial-consortium/bezier/issues/166](issue #166)
> [https://github.com/ros-industrial-consortium/bezier/pull/168](issue #168)
> [https://github.com/ros-industrial-consortium/bezier/issues/158](issue #158)
> [https://github.com/ros-industrial-consortium/bezier/issues/161](issue #161)

At your convenience, could you please spend some time helping me or give me some clue? It is really grateful!
Thanks again, and hope you have a nice day.
@GaoKui @VictorLamoine

VictorLamoine, 2020-09-29:
Can you debug the node with GDB please?
http://wiki.ros.org/roslaunch/Tutorials/Roslaunch%20Nodes%20in%20Valgrind%20or%20GDB

Don't forget to build the catkin workspace in debug mode!

```bash
catkin_make -DCMAKE_BUILD_TYPE=Debug
```

Please post the full `gdb` output here

AnsunWu, 2020-09-29:
Hello Victor @VictorLamoine,

Really grateful for your reply.
1. I tried to build the workspace in debug mode,  **it was compiled successfully.**

2. And then I debugged the node with GDB command `launch-prefix="gdb -ex run --args"`
The Rviz automatically showed up as shown below (**Some yellow ! showed in Rviz**):
[image]

Zooming in the mesh cad model:
[image]

**The 'backtrace' is shown as below:**

```ruby
ansonwu18@ansonwu18-virtual-machine:~/catkin_workspace/devel$ roslaunch bezier_examples fanuc_m10ia_surfacing.launch surfacing_mode:=true mesh_cad:=plane/plane_defect.ply
... logging to /home/ansonwu18/.ros/log/d3a98790-0253-11eb-9fec-000c29f16db5/roslaunch-ansonwu18-virtual-machine-66562.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://ansonwu18-virtual-machine:36401/

SUMMARY
========

PARAMETERS
 * /controller_joint_names: ['joint_1', 'join...
 * /mesh_cad_param: plane/plane_defec...
 * /mesh_defect_param: 
 * /move_group/allow_trajectory_execution: True
 * /move_group/capabilities: 
 * /move_group/controller_list: [{'joints': ['joi...
 * /move_group/disable_capabilities: 
 * /move_group/generic_hw_control_loop/cycle_time_error_threshold: 0.01
 * /move_group/generic_hw_control_loop/loop_hz: 300
 * /move_group/hardware_interface/joints: ['joint_1', 'join...
 * /move_group/hardware_interface/sim_control_mode: 1
 * /move_group/jiggle_fraction: 0.05
 * /move_group/joint_state_controller/publish_rate: 50
 * /move_group/joint_state_controller/type: joint_state_contr...
 * /move_group/manipulator/default_planner_config: LBKPIECE
 * /move_group/manipulator/longest_valid_segment_fraction: 0.005
 * /move_group/manipulator/planner_configs: ['SBL', 'EST', 'L...
 * /move_group/manipulator/projection_evaluator: joints(joint_1,jo...
 * /move_group/max_range: 5.0
 * /move_group/max_safe_path_cost: 1
 * /move_group/moveit_controller_manager: moveit_simple_con...
 * /move_group/moveit_manage_controllers: True
 * /move_group/moveit_sim_hw_interface/joint_model_group: controllers_initi...
 * /move_group/moveit_sim_hw_interface/joint_model_group_pose: controllers_initi...
 * /move_group/octomap_resolution: 0.025
 * /move_group/planner_configs/BFMT/balanced: 0
 * /move_group/planner_configs/BFMT/cache_cc: 1
 * /move_group/planner_configs/BFMT/extended_fmt: 1
 * /move_group/planner_configs/BFMT/heuristics: 1
 * /move_group/planner_configs/BFMT/nearest_k: 1
 * /move_group/planner_configs/BFMT/num_samples: 1000
 * /move_group/planner_configs/BFMT/optimality: 1
 * /move_group/planner_configs/BFMT/radius_multiplier: 1.0
 * /move_group/planner_configs/BFMT/type: geometric::BFMT
 * /move_group/planner_configs/BKPIECE/border_fraction: 0.9
 * /move_group/planner_configs/BKPIECE/failed_expansion_score_factor: 0.5
 * /move_group/planner_configs/BKPIECE/min_valid_path_fraction: 0.5
 * /move_group/planner_configs/BKPIECE/range: 0.0
 * /move_group/planner_configs/BKPIECE/type: geometric::BKPIECE
 * /move_group/planner_configs/BiEST/range: 0.0
 * /move_group/planner_configs/BiEST/type: geometric::BiEST
 * /move_group/planner_configs/BiTRRT/cost_threshold: 1e300
[... 273 lines cut ...]
    at /usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:652
#5  0x0000555555583c61 in Eigen::internal::generic_dense_assignment_kernel<Eigen::internal::evaluator<Eigen::Matrix<double, 4, 4, 0, 4, 4> >, Eigen::internal::evaluator<Eigen::Matrix<double, 4, 4, 0, 4, 4> >, Eigen::internal::assign_op<double, double>, 0>::assignPacketByOuterInner<32, 32, double __vector(4)>(long, long---Type <return> to continue, or q <return> to quit---return
) (this=0x7fffffffbe50, outer=0, inner=0)
    at /usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:666
#6  0x0000555555583976 in Eigen::internal::copy_using_evaluator_innervec_CompleteUnrolling<Eigen::internal::generic_dense_assignment_kernel<Eigen::internal::evaluator<Eigen::Matrix<double, 4, 4, 0, 4, 4> >, Eigen::internal::evaluator<Eigen::Matrix<double, 4, 4, 0, 4, 4> >, Eigen::internal::assign_op<double, double>, 0>, 0, 16>::run (kernel=...)
    at /usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:274
#7  0x00007ffff5007611 in rviz_visual_tools::RvizVisualTools::publishCylinder(Eigen::Matrix<double, 3, 1, 0, 3, 1> const&, Eigen::Matrix<double, 3, 1, 0, 3, 1> const&, std_msgs::ColorRGBA_<std::allocator<void> > const&, double, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) ()
   from /opt/ros/melodic/lib/librviz_visual_tools.so
#8  0x00007ffff50078bf in rviz_visual_tools::RvizVisualTools::publishCylinder(Eigen::Matrix<double, 3, 1, 0, 3, 1> const&, Eigen::Matrix<double, 3, 1, 0, 3, 1> const&, rviz_visual_tools::colors, double, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) ()
   from /opt/ros/melodic/lib/librviz_visual_tools.so
#9  0x00007ffff50084fb in rviz_visual_tools::RvizVisualTools::publishPath(std::vector<Eigen::Matrix<double, 3, 1, 0, 3, 1>, Eigen::aligned_allocator<Eigen::Matrix<double, 3, 1, 0, 3, 1> > > const&, rviz_visual_tools::colors, double, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) () from /opt/ros/melodic/lib/librviz_visual_tools.so
---Type <return> to continue, or q <return> to quit---return
[... cut ...]
#11 0x00007ffff7b99dc0 in BezierGrindingSurfacing::generateTrajectory[abi:cxx11](std::vector<Eigen::Transform<double, 3, 1, 0>, Eigen::aligned_allocator<Eigen::Transform<double, 3, 1, 0> > >&, std::vector<bool, std::allocator<bool> >&, bool) (this=0x555555cb2680, trajectory=std::vector of length 0, capacity 0, 
    is_grinding_pose=std::vector<bool> of length 0, capacity 0, 
    display_markers=true)
    at /home/ansonwu18/catkin_workspace/src/bezier/src/surfacing.cpp:441
#12 0x0000555555578909 in main (argc=1, argv=0x7fffffffdac8)
    at /home/ansonwu18/catkin_workspace/src/bezier_examples/bezier_examples/src/surfacing.cpp:104
(gdb) 

```

AnsunWu, 2020-09-29:
For the  **'Painting.launch'** in the 'Bezier example', the main issues are almost the same as the 'Surfacing .launch'

After launching in the GDB mode, the Rviz interface showed up as well, but with only **two 'yellow !'**, compared with the former surfacing project, the '**Marker**' item seems working normal, which is shown below:

[image]

The backtrace of this 'Painting project' is shown below:
```ruby
ansonwu18@ansonwu18-virtual-machine:~/catkin_workspace/devel$ roslaunch bezier_examples fanuc_m10ia_painting.launch mesh_cad:=ocean/ocean.ply
... logging to /home/ansonwu18/.ros/log/d36ac2f0-0256-11eb-9fec-000c29f16db5/roslaunch-ansonwu18-virtual-machine-67113.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://ansonwu18-virtual-machine:41599/

SUMMARY
========

PARAMETERS
 * /controller_joint_names: ['joint_1', 'join...
 * /mesh_cad_param: ocean/ocean.ply
 * /move_group/allow_trajectory_execution: True
 * /move_group/capabilities: 
 * /move_group/controller_list: [{'joints': ['joi...
 * /move_group/disable_capabilities: 
 * /move_group/generic_hw_control_loop/cycle_time_error_threshold: 0.01
 * /move_group/generic_hw_control_loop/loop_hz: 300
 * /move_group/hardware_interface/joints: ['joint_1', 'join...
 * /move_group/hardware_interface/sim_control_mode: 1
 * /move_group/jiggle_fraction: 0.05
 * /move_group/joint_state_controller/publish_rate: 50
 * /move_group/joint_state_controller/type: joint_state_contr...
 * /move_group/manipulator/default_planner_config: LBKPIECE
 * /move_group/manipulator/longest_valid_segment_fraction: 0.005
 * /move_group/manipulator/planner_configs: ['SBL', 'EST', 'L...
 * /move_group/manipulator/projection_evaluator: joints(joint_1,jo...
 * /move_group/max_range: 5.0
 * /move_group/max_safe_path_cost: 1
 * /move_group/moveit_controller_manager: moveit_simple_con...
 * /move_group/moveit_manage_controllers: True
 * /move_group/moveit_sim_hw_interface/joint_model_group: controllers_initi...
 * /move_group/moveit_sim_hw_interface/joint_model_group_pose: controllers_initi...
 * /move_group/octomap_resolution: 0.025
 * /move_group/planner_configs/BFMT/balanced: 0
 * /move_group/planner_configs/BFMT/cache_cc: 1
 * /move_group/planner_configs/BFMT/extended_fmt: 1
 * /move_group/planner_configs/BFMT/heuristics: 1
 * /move_group/planner_configs/BFMT/nearest_k: 1
 * /move_group/planner_configs/BFMT/num_samples: 1000
 * /move_group/planner_configs/BFMT/optimality: 1
 * /move_group/planner_configs/BFMT/radius_multiplier: 1.0
 * /move_group/planner_configs/BFMT/type: geometric::BFMT
 * /move_group/planner_configs/BKPIECE/border_fraction: 0.9
 * /move_group/planner_configs/BKPIECE/failed_expansion_score_factor: 0.5
 * /move_group/planner_configs/BKPIECE/min_valid_path_fraction: 0.5
 * /move_group/planner_configs/BKPIECE/range: 0.0
 * /move_group/planner_configs/BKPIECE/type: geometric::BKPIECE
 * /move_group/planner_configs/BiEST/range: 0.0
 * /move_group/planner_configs/BiEST/type: geometric::BiEST
 * /move_group/planner_configs/BiTRRT/cost_threshold: 1e300
 * /move_group/planner_configs/BiTRRT/frountier_node_ratio: 0.1
 * /move_group/planner_configs/BiTRRT/frountier_threshold: 0.0
 * /move_group/planner_configs/BiTRRT/init_temperature: 100
 * /move_group/planner_configs/BiTRRT/range: 0.0
 * /move_group/planner_configs/BiTRRT/temp_change_factor: 0.1
 * /move_group/planner_configs/BiTRRT/type: geometric::BiTRRT
 * /move_group/planner_configs/EST/goal_bias: 0.05
 * /move_group/planner_configs/EST/range: 0.0
[... 241 lines cut ...]
    at /usr/include/eigen3/Eigen/src/Core/arch/AVX/PacketMath.h:214
#2  0x00005555555895ba in Eigen::internal::ploadt<double __vector(4), 32>(Eigen::internal::unpacket_traits<double __vector(4)>::type const*) (from=0x7fffffffc2b0)
    at /usr/include/eigen3/Eigen/src/Core/GenericPacketMath.h:463
#3  Eigen::internal::evaluator<Eigen::PlainObjectBase<Eigen::Matrix<double, 4, 4, 0, 4, 4> > >::packet<32, double __vector(4)>(long, long) const (this=0x7fffffffc110, 
    row=0, col=0) at /usr/include/eigen3/Eigen/src/Core/CoreEvaluators.h:197
#4  0x000055555558939b in Eigen::internal::generic_dense_assignment_kernel<Eigen::internal::evaluator<Eigen::Matrix<double, 4, 4, 0, 4, 4> >, Eigen::internal::evaluator<Eigen::Matrix<double, 4, 4, 0, 4, 4> >, Eigen::internal::assign_op<double, double>, 0>::assignPacket<32, 32, double __vector(4)>(long, long) (this=0x7fffffffc130, 
    row=0, col=0) at /usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:652
#5  0x00005555555891ff in Eigen::internal::generic_dense_assignment_kernel<Eigen::internal::evaluator<Eigen::Matrix<double, 4, 4, 0, 4, 4> >, Eigen::internal::evaluator<Eigen::Matrix<double, 4, 4, 0, 4, 4> >, Eigen::internal::assign_op<double, double>, 0>::assignPacketByOuterInner<32, 32, double __vector(4)>(long, long) (
    this=0x7fffffffc130, outer=0, inner=0) at /usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:666
#6  0x0000555555588eee in Eigen::internal::copy_using_evaluator_innervec_CompleteUnrolling<Eigen::internal::generic_dense_assignment_kernel<Eigen::internal::evaluator<Eigen::Matrix<double, 4, 4, 0, 4, 4> >, Eigen::internal::evaluator<Eigen::Matrix<double, 4, 4, 0, 4, 4> >, Eigen::internal::assign_op<double, double>, 0>, 0, 16>::run (
[... cut ...]
#11 0x00007ffff7bb389a in BezierPainting::generateTrajectory[abi:cxx11](std::vector<Eigen::Transform<double, 3, 1, 0>, Eigen::aligned_allocator<Eigen::Transform<double, 3, 1, 0> > >&, std::vector<bool, std::allocator<bool> >&, bool) (this=0x555555cbb640, trajectory=std::vector of length 1055, capacity 2048 = {...}, 
    is_grinding_pose=std::vector<bool> of length 1056, capacity 2048 = {...}, display_markers=true) at /home/ansonwu18/catkin_workspace/src/bezier/src/painting.cpp:152
#12 0x000055555557d394 in main (argc=1, argv=0x7fffffffdac8) at /home/ansonwu18/catkin_workspace/src/bezier_examples/bezier_examples/src/painting.cpp:94
(gdb) 

```

AnsunWu, 2020-09-29:
Thanks again @VictorLamoine , really grateful for your help, merci.

VictorLamoine, 2020-09-29:
Thanks for the backtrace. Looks like this bug is https://eigen.tuxfamily.org/bz/show_bug.cgi?id=1427

What Eigen version are you using?
```bash
dpkg -l *eigen*
```

AnsunWu, 2020-09-29:
Hello Victor @VictorLamoine 

The Eigen version is 3.3.4-4 as in the screenshot below;
By the way, the backtrace is from the **ROS Melodic in Ubuntu 18.04:**

[image]

Thanks again, Victor.

VictorLamoine, 2020-09-29:
Install the latest version (http://eigen.tuxfamily.org/index.php?title=Main_Page) and configure CMake to use this new version.
Compile and test again, it should solve the problem as this seems to be an Eigen problem.

AnsunWu, 2020-09-30:
> Install the latest version (http://eigen.tuxfamily.org/index.php?title=Main_Page) and configure CMake to use this new version.
> Compile and test again, it should solve the problem as this seems to be an Eigen problem.

Hello Victor,

Thanks for your reply, I will try to do in this way and give you the feedback soon.
Really thanks for your help.

AnsunWu, 2020-10-01:
Hello Victor @VictorLamoine 

I uninstalled the old version **Eigen (3.3.4-4)** and installed the latest version **Eigen (3.3.7).**
And recompiled the workspace by using the latest Eigen, compile successfully.

**Unfortunately the crash issue still exists...**
**But the good news is that the backtrace is not the same as the former one:**
I tried to search the reason for the crash on google and nearly all the forums, but still get no luck..

Could you please help me to check the backtrace below? 
Really thanks for your help and time.

```ruby
setting /run_id to e8627b86-0408-11eb-93c0-000c29f16db5
process[rosout-1]: started with pid [14408]
started core service [/rosout]
process[industrial_robot_simulator-2]: started with pid [14415]
process[joint_trajectory_action-3]: started with pid [14416]
process[robot_state_publisher-4]: started with pid [14417]
process[move_group-5]: started with pid [14423]
process[bezier_examples_surfacing-6]: started with pid [14429]
process[rviz-7]: started with pid [14432]
[ INFO] [1601572188.316673267]: Loading robot model 'fanuc_m10ia'...
[ INFO] [1601572188.317210896]: No root/virtual joint specified in SRDF. Assuming fixed joint
GNU gdb (Ubuntu 8.1-0ubuntu3.2) 8.1.0.20180409-git
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/ansonwu18/catkin_workspace/devel/lib/bezier_examples/bezier_examples_surfacing...(no debugging symbols found)...done.
Starting program: /home/ansonwu18/catkin_workspace/devel/lib/bezier_examples/bezier_examples_surfacing __name:=bezier_examples_surfacing __log:=/home/ansonwu18/.ros/log/e8627b86-0408-11eb-93c0-000c29f16db5/bezier_examples_surfacing-6.log
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[ WARN] [1601572188.355995849]: Kinematics solver doesn't support #attempts anymore, but only a timeout.
Please remove the parameter '/robot_description_kinematics/manipulator/kinematics_solver_attempts' from your configuration.
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[ INFO] [1601572188.392782372]: Publishing maintained planning scene on 'monitored_planning_scene'
[ INFO] [1601572188.394462228]: MoveGroup debug mode is OFF
Starting planning scene monitors...
[ INFO] [1601572188.394490565]: Starting planning scene monitor
[ INFO] [1601572188.395809804]: Listening to '/planning_scene'
[ INFO] [1601572188.395850965]: Starting world geometry update monitor for collision objects, attached objects, octomap updates.
[ INFO] [1601572188.397132669]: Listening to '/collision_object'
[ INFO] [1601572188.398824393]: Listening to '/planning_scene_world' for planning scene world geometry
[ERROR] [1601572188.399395747]: Failed to find 3D sensor plugin parameters for octomap generation
[ INFO] [1601572188.405607297]: Listening to '/attached_collision_object' for attached collision objects
Planning scene monitors started.
[ INFO] [1601572188.423867820]: Initializing OMPL interface using ROS parameters
[ INFO] [1601572188.442083848]: Using planning interface 'OMPL'
[ INFO] [1601572188.444474398]: Param 'default_workspace_bounds' was not set. Using default value: 10
[ INFO] [1601572188.445061105]: Param 'start_state_max_bounds_error' was set to 0.1
[ INFO] [1601572188.445603880]: Param 'start_state_max_dt' was not set. Using default value: 0.5
[ INFO] [1601572188.446068123]: Param 'start_state_max_dt' was not set. Using default value: 0.5
[ INFO] [1601572188.446562185]: Param 'jiggle_fraction' was set to 0.05
[ INFO] [1601572188.447006328]: Param 'max_sampling_attempts' was not set. Using default value: 100
[ INFO] [1601572188.447080861]: Using planning request adapter 'Add Time Parameterization'
[ INFO] [1601572188.447135041]: Using planning request adapter 'Fix Workspace Bounds'
[ INFO] [1601572188.447178956]: Using planning request adapter 'Fix Start State Bounds'
[ INFO] [1601572188.447222009]: Using planning request adapter 'Fix Start State In Collision'
[ INFO] [1601572188.447265255]: Using planning request adapter 'Fix Start State Path Constraints'
[ERROR] [1601572188.451749939]: name, joints, action_ns, and type must be specifed for each controller
[ INFO] [1601572188.451859289]: Returned 0 controllers in list
[... 56 lines cut ...]
[ WARN] [1601572205.678835577]: Bezier::keepUpperPartofDilatedMesh: Computing normals...
[ INFO] [1601572207.698613764]: BezierGrindingSurfacing::generateTrajectory: Generating grinding trajectories
[ INFO] [1601572207.726374599]: BezierGrindingSurfacing::generateTrajectory: Generating extrication trajectories

Thread 1 "bezier_examples" received signal SIGSEGV, Segmentation fault.
0x000055555557b58f in double __vector(4) Eigen::internal::pload<double __vector(4)>(Eigen::internal::unpacket_traits<double __vector(4)>::type const*) ()
(gdb) bt
[... cut ...]
#8  0x00007ffff5edbbf9 in Bezier::displayTrajectory(std::vector<Eigen::Transform<double, 3, 1, 0>, Eigen::aligned_allocator<Eigen::Transform<double, 3, 1, 0> > > const&, rviz_visual_tools::colors, bool, bool) ()
   from /home/ansonwu18/catkin_workspace/devel/lib/libbezier.so
#9  0x00007ffff7b9a2e0 in BezierGrindingSurfacing::generateTrajectory[abi:cxx11](std::vector<Eigen::Transform<double, 3, 1, 0>, Eigen::aligned_allocator<Eigen::Transform<double, 3, 1, 0> > >&, std::vector<bool, std::allocator<bool> >&, bool) () from /home/ansonwu18/catkin_workspace/devel/lib/libbezier_surfacing.so
#10 0x0000555555578909 in main ()
(gdb) 

```

VictorLamoine, 2020-10-02:
You didn't need to uninstall the Eigen 3.3.4-4 but anyway, the problem lies in `rviz_visual_tools`, not in `bezier`.

Are you using the source version of `rviz_visual_tools` or the released package?
If you are using the released package clone https://github.com/PickNikRobotics/rviz_visual_tools in your catkin workspace and try again.

AnsunWu, 2020-10-03:
Hello Victor @VictorLamoine 

Thanks for your reply and help.
In the former test, the source version of the  **rviz_visual_tools**  was used.

Today, I cloned the **released package** from the GitHub link [https://github.com/PickNikRobotics/rviz_visual_tools](url)
and recompiled the workspace successfully.

The issue of **rviz_visual_tools** seems fixed, but the **process died issue** is still existed.
The latest backtrace is shown below, 5 items left. **Maybe the issues are related to VTK and PCL libraries.**

Thanks again for your help and time.

```ruby
*******************************************************
* MoveGroup using: 
*     - ApplyPlanningSceneService
*     - ClearOctomapService
*     - CartesianPathService
*     - ExecuteTrajectoryAction
*     - GetPlanningSceneService
*     - KinematicsService
*     - MoveAction
*     - PickPlaceAction
*     - MotionPlanService
*     - QueryPlannersService
*     - StateValidationService
********************************************************

[ INFO] [1601684652.811018251]: MoveGroup context using planning plugin ompl_interface/OMPLPlanner
[ INFO] [1601684652.811046290]: MoveGroup context initialization complete

You can start planning now!

[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[New Thread 0x7fff9fd21700 (LWP 24393)]
[New Thread 0x7fff9f520700 (LWP 24394)]
[New Thread 0x7fff9ed1f700 (LWP 24395)]
[New Thread 0x7fff9e51e700 (LWP 24396)]
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[ INFO] [1601684675.440831557]: CAD mesh file:plane/plane_defect.ply
[ INFO] [1601684675.442097565]: Bezier::Bezier: RViz visualization tool is initialized in 'base' and the topic name is 'rviz_visual_tools'
[ WARN] [1601684675.445440537]: Waiting for a subscriber on topic "/bezier_cad_mesh"
[ INFO] [1601684676.449576618]: Topic '/rviz_visual_tools' waiting for subscriber...

Thread 1 "bezier_examples" received signal SIGSEGV, Segmentation fault.
0x00007fffecca93ba in vtkPointSet::SetPoints(vtkPoints*) [clone .localalias.36]
    () from /usr/local/lib/libvtkCommonDataModel-8.2.so.1
(gdb) 
(gdb) bt
#0  0x00007fffecca93ba in vtkPointSet::SetPoints(vtkPoints*) [clone .localalias.36] () 
   from /usr/local/lib/libvtkCommonDataModel-8.2.so.1
#1  0x00007ffff0ad76b9 in pcl::VTKUtils::vtk2mesh(vtkSmartPointer<vtkPolyData> const&, pcl::PolygonMesh&) ()
   from /usr/lib/x86_64-linux-gnu/libpcl_surface.so.1.8
#2  0x00007fffcba8c178 in Bezier::estimateGlobalMeshNormal(vtkSmartPointer<vtkPolyData>&, Eigen::Matrix<double, 3, 1, 0, 3, 1>&, unsigned int) ()
   from /home/ansonwu18/catkin_workspace/build/devel/lib/libbezier.so
#3  0x00007fffcba8cb79 in Bezier::estimateSlicingOrientation(vtkSmartPointer<vtkPolyData>&, Eigen::Matrix<double, 3, 1, 0, 3, 1>&, Eigen::Matrix<double, 3, 1, 0, 3, 1>&) () from /home/ansonwu18/catkin_workspace/build/devel/lib/libbezier.so
#4  0x00007ffff772b923 in BezierGrindingSurfacing::generateTrajectory[abi:cxx11](std::vector<Eigen::Transform<double, 3, 1, 0>, Eigen::aligned_allocator<Eigen::Transform<double, 3, 1, 0> > >&, std::vector<bool, std::allocator<bool> >&, bool) ()
   from /home/ansonwu18/catkin_workspace/build/devel/lib/libbezier_surfacing.so
#5  0x0000000000428d66 in main ()
(gdb) 

```

VictorLamoine, 2020-10-05:
Now the problem lies in the PCL `pcl::VTKUtils::vtk2mesh` function. Build PCL in debug mode and try to determine what is making it crash

PR #167 (2018-12-21): Return error if stripper has more than one line
Fixes #166