ros-industrial-attic/bezier #169: Move application(s) to separate repository

nlimpert, 2019-02-13:
Currently the `bezier_application` package has a dependency on `fanuc_m20ia_support` which obviously requires anyone willing to use this metapackage to fullfill this dependency.
I would be in favor of a solution that keeps the `bezier_library` in a kind-of standalone shape so I would propose to move the `bezier_application` package to a separate repository and in turn keep the fanuc application as an example of how to make use of the `bezier_library`.

If you agree I can create a PR for this.

VictorLamoine, 2019-02-13:
I agree, Bezier was one my first ROS package, and it really looks ugly now :)
I'm going to ask for someone to create the `bezier_application` repository because I don't have the rights to do that.

I'll keep you updated when it's created.

gavanderhoorn, 2019-02-13:
@VictorLamoine: is `fanuc_m20ia_support` really a `build_depend`? It's listed as a `depend` in the manifest, which would imply it's both a run as well as build dependency.

If it's only needed to launch `.launch` files, it should be an `exec_depend` and could be ignored by users if they're not interested in using the `.lauch` files that use `fanuc_m20ia_support`.

VictorLamoine, 2019-02-13:
No it's not a build depend, that's exactly why bezier is ugly :)
Anyway I think the applications should be split from the library itself.

The application package is just an example on how to use the package.

gavanderhoorn, 2019-02-13:
> No it's not a build depend, that's exactly why bezier is ugly :)

Then it would make sense to fix that in any case.

> Anyway I think the applications should be split from the library itself.
> 
> The application package is just an example on how to use the package.

I'm happy to create a new repository, but depending on how users setup their workspace, that might not solve the issue here: if they clone both this repository and the new one (which would seem like a thing they would do, as they'd want to have an example that shows how to use the library), they run into the same problem.

Unfortunately there is no support for optional dependencies in ROS pkg manifests (v2 in any case).

nlimpert, 2019-02-13:
I agree to @gavanderhoorn. However the idea was to have them both separated. I'm perfectly fine with an example package defining a proper dependency on a certain robot. That is being an example package and not the one that a person would use in production but to learn how to setup his own robot using the `bezier_library`.

If a user wants to setup a clean workspace for one robot other than `fanuc_mia20` he can't get rid of these dependencies currently, but he would if the packages would be split up in two repositories.

VictorLamoine, 2019-02-13:
Agreed, the package need to be fixed and be properly written. I've come a long way since bezier, now I use catkin_lint to help me build cleaner packages and it works great!

Please create a new repository and we will do some cleaning in both the two repositories

gavanderhoorn, 2019-02-13:
I'm actually suggesting to either a) remove the `depend` on `fanuc_m20ia_support` from `bezier_application` and clearly state that in order to use the *example launch files* users need to install that package manually, or b) convert the `depend` into the proper `exec_depend` and update the `rosdep install ..` line shown with a `--skip-keys="fanuc_m20ia_support"`.

@VictorLamoine if you want a new repository I can create it though.

VictorLamoine, 2019-02-13:
I will create a wstool rosinstall filee and proper instructions on how to create a complete workspace with all the dependencies.

Yes please

nlimpert, 2019-02-13:
> I'm actually suggesting to either a) remove the `depend` on `fanuc_m20ia_support` from `bezier_application` and clearly state that in order to use the _example launch files_ users need to install [...]

Well I guess in order to keep `rosdep` from installing these dependencies (in terms of `rosdep` being used in it's normal fashion without caring additionally for a certain package) _a)_ would be the only viable - if @VictorLamoine would prefer to keep the application inside this repository.

gavanderhoorn, 2019-02-13:
@nlimpert: it's academic at this point, but `--skip-keys=""` is exactly meant to allow you to ask `rosdep` to ignore certain dependencies. That would make b possible as well.

gavanderhoorn, 2019-02-13:
@VictorLamoine: what about `bezier_examples`, instead of `bezier_application`? I like that name, as it clearly conveys what the contents of the repository is.

nlimpert, 2019-02-13:
> @nlimpert: it's academic at this point, but `--skip-keys=""` is exactly meant to allow you to ask `rosdep` to ignore certain dependencies. That would make b possible as well.

Yes I got that point - but I guess the user would have to always keep in mind that there's this certain package to be omitted when using `rosdep` for one complete workspace, right?
(That's at least what I meant)

VictorLamoine, 2019-02-14:
Closing as I moved the code to the new repository. Now I will do some cleaning

closed 2019-02-14 by VictorLamoine

VictorLamoine, 2019-02-18:
@nlimpert I've done some cleaning and updated everything to melodic.

Robot does not move in the example because of a missing piece in the MoveIt configuration I have generated. I don't have time to fix this but if you have please open a pull request (https://github.com/ros-industrial-consortium/bezier_examples/issues/2).

nlimpert, 2019-02-18:
Thank you for the cleanup!
I'm not sure when I can find the time to have a look but I'll see if I can help asap.