ros-industrial-attic/bezier #174: Compiling error

DanielSchimpf, 2019-05-29:
Hello,

I am running into some warnings and errors when trying to compile the package. 

I am using Ubuntu 16 and ROS Kinetic, because bezier is listed as an dependency in the fanuc_grinding package and there Ubuntu 16 with ROS Kinetic was used.

 I made a fresh installation of Ubuntu and ROS Kinetic. The only other things that I installed in addition to the things mentioned in the install instructions for bezier are PCL 1.8, rviz-visual-tools, libproj-dev and python-catkin-tools.

PCL 1.8, libproj-dev and rviz-visual-tools I installed because of earlier error messages when trying to compile bezier. However, now I am a little stuck. It seems to be an issue with vtk.

I would be very thankful for some help!

Let me know if there is would be any other information I should provide.

Thanks!!!

Daniel 

The compilation output is pasted below (its long, sorry...):

dan@dan-VirtualBox:~/catkin_ws$ catkin build
--------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/kinetic
Workspace:                   /home/dan/catkin_ws
--------------------------------------------------------
Build Space:        [exists] /home/dan/catkin_ws/build
Devel Space:        [exists] /home/dan/catkin_ws/devel
Install Space:      [unused] /home/dan/catkin_ws/install
Log Space:          [exists] /home/dan/catkin_ws/logs
Source Space:       [exists] /home/dan/catkin_ws/src
DESTDIR:            [unused] None
--------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
--------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
--------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
--------------------------------------------------------
Workspace configuration appears valid.
--------------------------------------------------------
[build] Found '1' packages in 0.0 seconds.                                     
[build] Package table is up to date.                                           
Starting  >>> bezier                                                           
_______________________________________________________________________________
Errors     << bezier:make /home/dan/catkin_ws/logs/bezier/build.make.002.log   
cc1plus: warning: /usr/local/lib/cmake/vtk-8.2/UseVTK.cmake: not a directory
In file included from /home/dan/catkin_ws/src/bezier/include/bezier/bezier_library.hpp:36:0,
                 from /home/dan/catkin_ws/src/bezier/src/bezier_library.cpp:1:
/usr/include/vtk-6.2/vtkMath.h:672:3: warning: multi-line comment [-Wcomment]
   // a & b \\
   ^
/usr/include/vtk-6.2/vtkMath.h:676:3: warning: multi-line comment [-Wcomment]
   // 1 & 0 \\
   ^
/usr/include/vtk-6.2/vtkMath.h:679:3: warning: multi-line comment [-Wcomment]
   // a & b \\
   ^
/home/dan/catkin_ws/src/bezier/src/bezier_library.cpp: In member function ‘bool Bezier::appendInputMesh(std::__cxx11::string)’:
/home/dan/catkin_ws/src/bezier/src/bezier_library.cpp:23:19: error: ‘vtkAbstractPolyDataReader’ was not declared in this scope
   vtkSmartPointer<vtkAbstractPolyDataReader> reader;
                   ^
/home/dan/catkin_ws/src/bezier/src/bezier_library.cpp:23:44: error: template argument 1 is invalid
   vtkSmartPointer<vtkAbstractPolyDataReader> reader;
[... 197 lines cut ...]
     return vtkSmartPointer<T>(T::New(), NoReference());
                                     ^
/usr/include/vtk-6.2/vtkSmartPointer.h: In instantiation of ‘vtkSmartPointer<T>::operator T*() const [with T = vtkIdList]’:
/home/dan/catkin_ws/src/bezier/src/bezier_library.cpp:1094:81:   required from here
/usr/include/vtk-6.2/vtkSmartPointer.h:79:40: error: invalid static_cast from type ‘vtkObjectBase* const’ to type ‘vtkIdList*’
     return static_cast<T*>(this->Object);
                                        ^
/usr/include/vtk-6.2/vtkSmartPointer.h: In instantiation of ‘T* vtkSmartPointer<T>::operator->() const [with T = vtkIdList]’:
/home/dan/catkin_ws/src/bezier/src/bezier_library.cpp:1105:48:   required from here
/usr/include/vtk-6.2/vtkSmartPointer.h:94:40: error: invalid static_cast from type ‘vtkObjectBase* const’ to type ‘vtkIdList*’
     return static_cast<T*>(this->Object);
                                        ^
make[2]: *** [CMakeFiles/bezier.dir/src/bezier_library.cpp.o] Error 1
make[1]: *** [CMakeFiles/bezier.dir/all] Error 2
make: *** [all] Error 2
cd /home/dan/catkin_ws/build/bezier; catkin build --get-env bezier | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...............................................................................
Failed     << bezier:make           [ Exited with code 2 ]                     
Failed    <<< bezier                [ 12.2 seconds ]                           
[build] Summary: 0 of 1 packages succeeded.                                    
[build]   Ignored:   None.                                                     
[build]   Warnings:  None.                                                     
[build]   Abandoned: None.                                                     
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 12.3 seconds total.

VictorLamoine, 2019-06-03:
`bezier` compiles/works as it should on Melodic so feel free to update to Ubuntu 18.04 and ROS Melodic.

fanuc_grinding is not maintained anymore and pretty poor quality code overall (my bad :stuck_out_tongue_closed_eyes:). 
I do not recommend trying to compile/run it. The biggest problem is that fanuc_post_processor changed A LOT so you have to use an old version, something around this commit should work: https://gitlab.com/InstitutMaupertuis/fanuc_post_processor/commit/5c1d0545d7320f5ca89f3245a0508b6fdda682e6

I experience the same compilation issue. I'll take a look and publish a fix soon.
It seems VTK 6 includes are used instead of VTK 8.

VictorLamoine, 2019-06-03:
Fixed with https://github.com/ros-industrial-consortium/bezier/commit/fafe79d6ef974bbb8050a8b0333cd9399bb2f8c0.

Please tell me if if works for you. If it doesn't re-open the issue !

DanielSchimpf, 2019-06-06:
Thank you! 
Sadly that does not seem to resolve the issue. Compilation output seems to be the same/very similar:
dan@dan-VirtualBox:~/catkin_ws$ catkin build
--------------------------------------------------------
Profile:                     default
Extending:             [env] /opt/ros/kinetic
Workspace:                   /home/dan/catkin_ws
--------------------------------------------------------
Build Space:        [exists] /home/dan/catkin_ws/build
Devel Space:        [exists] /home/dan/catkin_ws/devel
Install Space:      [unused] /home/dan/catkin_ws/install
Log Space:          [exists] /home/dan/catkin_ws/logs
Source Space:       [exists] /home/dan/catkin_ws/src
DESTDIR:            [unused] None
--------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
--------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
--------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
--------------------------------------------------------
Workspace configuration appears valid.
--------------------------------------------------------
[build] Found '2' packages in 0.0 seconds.                                     
[build] Updating package table.                                                
Warning: generated devel space setup files have been deleted.
Starting  >>> catkin_tools_prebuild                                            
Finished  <<< catkin_tools_prebuild                [ 13.6 seconds ]            
Starting  >>> beginner_tutorials                                               
Finished  <<< beginner_tutorials                   [ 1 minute and 31.0 seconds ]
Starting  >>> bezier                                                           
_______________________________________________________________________________
Warnings   << bezier:cmake /home/dan/catkin_ws/logs/bezier/build.cmake.005.log 
** WARNING ** io features related to openni2 will be disabled
** WARNING ** io features related to ensenso will be disabled
** WARNING ** io features related to davidSDK will be disabled
** WARNING ** io features related to dssdk will be disabled
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
** WARNING ** io features related to libusb-1.0 will be disabled
** WARNING ** visualization features related to openni2 will be disabled
** WARNING ** visualization features related to ensenso will be disabled
** WARNING ** visualization features related to davidSDK will be disabled
** WARNING ** visualization features related to dssdk will be disabled
** WARNING ** visualization features related to rssdk will be disabled
cd /home/dan/catkin_ws/build/bezier; catkin build --get-env bezier | catkin env -si  /usr/bin/cmake /home/dan/catkin_ws/src/bezier --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/dan/catkin_ws/devel/.private/bezier -DCMAKE_INSTALL_PREFIX=/home/dan/catkin_ws/install; cd -
...............................................................................
_______________________________________________________________________________
Errors     << bezier:make /home/dan/catkin_ws/logs/bezier/build.make.005.log   
In file included from /home/dan/catkin_ws/src/bezier/include/bezier/bezier_library.hpp:36:0,
                 from /home/dan/catkin_ws/src/bezier/src/bezier_library.cpp:1:
/usr/include/vtk-6.2/vtkMath.h:672:3: warning: multi-line comment [-Wcomment]
   // a & b \\
   ^
/usr/include/vtk-6.2/vtkMath.h:676:3: warning: multi-line comment [-Wcomment]
   // 1 & 0 \\
   ^
/usr/include/vtk-6.2/vtkMath.h:679:3: warning: multi-line comment [-Wcomment]
   // a & b \\
   ^
/home/dan/catkin_ws/src/bezier/src/bezier_library.cpp: In member function ‘bool Bezier::appendInputMesh(std::__cxx11::string)’:
/home/dan/catkin_ws/src/bezier/src/bezier_library.cpp:23:19: error: ‘vtkAbstractPolyDataReader’ was not declared in this scope
   vtkSmartPointer<vtkAbstractPolyDataReader> reader;
                   ^
[... 199 lines cut ...]
     return vtkSmartPointer<T>(T::New(), NoReference());
                                     ^
/usr/include/vtk-6.2/vtkSmartPointer.h: In instantiation of ‘vtkSmartPointer<T>::operator T*() const [with T = vtkIdList]’:
/home/dan/catkin_ws/src/bezier/src/bezier_library.cpp:1094:81:   required from here
/usr/include/vtk-6.2/vtkSmartPointer.h:79:40: error: invalid static_cast from type ‘vtkObjectBase* const’ to type ‘vtkIdList*’
     return static_cast<T*>(this->Object);
                                        ^
/usr/include/vtk-6.2/vtkSmartPointer.h: In instantiation of ‘T* vtkSmartPointer<T>::operator->() const [with T = vtkIdList]’:
/home/dan/catkin_ws/src/bezier/src/bezier_library.cpp:1105:48:   required from here
/usr/include/vtk-6.2/vtkSmartPointer.h:94:40: error: invalid static_cast from type ‘vtkObjectBase* const’ to type ‘vtkIdList*’
     return static_cast<T*>(this->Object);
                                        ^
make[2]: *** [CMakeFiles/bezier.dir/src/bezier_library.cpp.o] Error 1
make[1]: *** [CMakeFiles/bezier.dir/all] Error 2
make: *** [all] Error 2
cd /home/dan/catkin_ws/build/bezier; catkin build --get-env bezier | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
[... cut ...]
Failed    <<< bezier                               [ 1 minute and 12.4 seconds ]
[build] Summary: 2 of 3 packages succeeded.                                    
[build]   Ignored:   None.                                                     
[build]   Warnings:  1 packages succeeded with warnings.                       
[build]   Abandoned: None.                                                     
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 2 minutes and 57.4 seconds total.

VictorLamoine, 2019-06-07:
Have you reconfigured the project ? CMake stores some variables and it did store the path to VTK.
Simplest solution is to remove the `build` and `devel` directories and run `catkin_make` again.

Tell me if it works for you!

DanielSchimpf, 2019-06-07:
Yeah I deleted both build and devel folder before trying to compile. Tried it again with `catkin build` and`catkin_make`, both failed. (Does it ever happen that one of them works but not the other[Assuming `build` and `devel` directories were deleted before]?)

<details>
  <summary>compilation log catkin_make</summary>

```
dan@dan-VirtualBox:~/catkin_ws$ catkin_make
Base path: /home/dan/catkin_ws
Source space: /home/dan/catkin_ws/src
Build space: /home/dan/catkin_ws/build
Devel space: /home/dan/catkin_ws/devel
Install space: /home/dan/catkin_ws/install
####
#### Running command: "cmake /home/dan/catkin_ws/src -DCATKIN_DEVEL_PREFIX=/home/dan/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/dan/catkin_ws/install -G Unix Makefiles" in "/home/dan/catkin_ws/build"
####
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/dan/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/dan/catkin_ws/devel;/opt/ros/kinetic
-- This workspace overlays: /home/dan/catkin_ws/devel;/opt/ros/kinetic
-- Found PythonInterp: /usr/bin/python (found version "2.7.12") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/dan/catkin_ws/build/test_results
-- Found gmock sources under '/usr/src/gmock': gmock will be built
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found gtest sources under '/usr/src/gmock': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.14
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 2 packages in topological order:
-- ~~  - beginner_tutorials
-- ~~  - bezier
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'beginner_tutorials'
-- ==> add_subdirectory(beginner_tutorials)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- beginner_tutorials: 1 messages, 1 services
-- +++ processing catkin package: 'bezier'
-- ==> add_subdirectory(bezier)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Checking for module 'eigen3'
--   Found eigen3, version 3.2.92
-- Found eigen: /usr/include/eigen3  
-- Boost version: 1.58.0
-- Found the following Boost libraries:
[... 479 lines cut ...]
                                     ^
/usr/include/vtk-6.2/vtkSmartPointer.h: In instantiation of ‘vtkSmartPointer<T>::operator T*() const [with T = vtkIdList]’:
/home/dan/catkin_ws/src/bezier/src/bezier_library.cpp:1094:81:   required from here
/usr/include/vtk-6.2/vtkSmartPointer.h:79:40: error: invalid static_cast from type ‘vtkObjectBase* const’ to type ‘vtkIdList*’
     return static_cast<T*>(this->Object);
                                        ^
/usr/include/vtk-6.2/vtkSmartPointer.h: In instantiation of ‘T* vtkSmartPointer<T>::operator->() const [with T = vtkIdList]’:
/home/dan/catkin_ws/src/bezier/src/bezier_library.cpp:1105:48:   required from here
/usr/include/vtk-6.2/vtkSmartPointer.h:94:40: error: invalid static_cast from type ‘vtkObjectBase* const’ to type ‘vtkIdList*’
     return static_cast<T*>(this->Object);
                                        ^
bezier/CMakeFiles/bezier.dir/build.make:62: recipe for target 'bezier/CMakeFiles/bezier.dir/src/bezier_library.cpp.o' failed
make[2]: *** [bezier/CMakeFiles/bezier.dir/src/bezier_library.cpp.o] Error 1
CMakeFiles/Makefile2:3402: recipe for target 'bezier/CMakeFiles/bezier.dir/all' failed
make[1]: *** [bezier/CMakeFiles/bezier.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 55%] Linking CXX executable /home/dan/catkin_ws/devel/lib/beginner_tutorials/listener
[ 55%] Built target listener
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j2 -l2" failed

 ```

</details>

DanielSchimpf, 2019-06-07:
I also tried compiling it in Ubuntu 18 with Melodic. It produces a VTK error as well. Do I need to install the PCL or anything else to be able to compile it?

Thanks, I really appreciate your help!

<details>
  <summary>compilation log catkin build</summary>

```
dan@dan-VirtualBox:~/catkin_ws$ catkin build
--------------------------------------------------------
Profile:                     default
Extending:             [env] /opt/ros/melodic
Workspace:                   /home/dan/catkin_ws
--------------------------------------------------------
Build Space:        [exists] /home/dan/catkin_ws/build
Devel Space:        [exists] /home/dan/catkin_ws/devel
Install Space:      [unused] /home/dan/catkin_ws/install
Log Space:         [missing] /home/dan/catkin_ws/logs
Source Space:       [exists] /home/dan/catkin_ws/src
DESTDIR:            [unused] None
--------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
--------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
--------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
--------------------------------------------------------
Workspace configuration appears valid.
--------------------------------------------------------
[build] Found '1' packages in 0.0 seconds.                                     
[build] Updating package table.                                                
Warning: generated devel space setup files have been deleted.
Starting  >>> catkin_tools_prebuild                                            
Finished  <<< catkin_tools_prebuild                [ 3.8 seconds ]             
Starting  >>> bezier                                                           
_______________________________________________________________________________
Warnings   << bezier:cmake /home/dan/catkin_ws/logs/bezier/build.cmake.000.log 
** WARNING ** io features related to ensenso will be disabled
** WARNING ** io features related to davidSDK will be disabled
** WARNING ** io features related to dssdk will be disabled
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
** WARNING ** io features related to libusb-1.0 will be disabled
** WARNING ** visualization features related to ensenso will be disabled
** WARNING ** visualization features related to davidSDK will be disabled
** WARNING ** visualization features related to dssdk will be disabled
** WARNING ** visualization features related to rssdk will be disabled
cd /home/dan/catkin_ws/build/bezier; catkin build --get-env bezier | catkin env -si  /usr/bin/cmake /home/dan/catkin_ws/src/bezier --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/dan/catkin_ws/devel/.private/bezier -DCMAKE_INSTALL_PREFIX=/home/dan/catkin_ws/install; cd -
...............................................................................
_______________________________________________________________________________
Errors     << bezier:make /home/dan/catkin_ws/logs/bezier/build.make.000.log   
In file included from /home/dan/catkin_ws/src/bezier/include/bezier/bezier_library.hpp:36:0,
                 from /home/dan/catkin_ws/src/bezier/src/bezier_library.cpp:1:
/usr/include/vtk-6.3/vtkMath.h:664:3: warning: multi-line comment [-Wcomment]
   // a & b \\
   ^
/usr/include/vtk-6.3/vtkMath.h:668:3: warning: multi-line comment [-Wcomment]
   // 1 & 0 \\
   ^
/usr/include/vtk-6.3/vtkMath.h:671:3: warning: multi-line comment [-Wcomment]
   // a & b \\
   ^
/home/dan/catkin_ws/src/bezier/src/bezier_library.cpp: In member function ‘bool Bezier::sliceMeshWithPlane(const vtkSmartPointer<vtkPolyData>&, const Vector4d&, const Vector3d&, vtkSmartPointer<vtkStripper>&)’:
/home/dan/catkin_ws/src/bezier/src/bezier_library.cpp:461:13: error: ‘class vtkStripper’ has no member named ‘JoinContiguousSegmentsOn’
[... 156 lines cut ...]
/usr/include/vtk-6.3/vtkSmartPointer.h:79:12: error: invalid static_cast from type ‘vtkObjectBase* const’ to type ‘vtkIdList*’
     return static_cast<T*>(this->Object);
            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/vtk-6.3/vtkSmartPointer.h: In instantiation of ‘T* vtkSmartPointer<T>::operator->() const [with T = vtkIdList]’:
/home/dan/catkin_ws/src/bezier/src/bezier_library.cpp:1105:48:   required from here
/usr/include/vtk-6.3/vtkSmartPointer.h:94:12: error: invalid static_cast from type ‘vtkObjectBase* const’ to type ‘vtkIdList*’
     return static_cast<T*>(this->Object);
            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/bezier.dir/src/bezier_library.cpp.o] Error 1
make[1]: *** [CMakeFiles/bezier.dir/all] Error 2
make: *** [all] Error 2
cd /home/dan/catkin_ws/build/bezier; catkin build --get-env bezier | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...............................................................................
Failed     << bezier:make                          [ Exited with code 2 ]      
Failed    <<< bezier                               [ 20.7 seconds ]            
[build] Summary: 1 of 2 packages succeeded.                                    
[build]   Ignored:   None.                                                     
[build]   Warnings:  1 packages succeeded with warnings.                       
[build]   Abandoned: None.                                                     
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 24.5 seconds total. 

 ```

</details>

VictorLamoine, 2019-06-11:
I can't reproduce the problem now. Can you remove the `devel` / `build` directories and try again? I think something was cached in CMake.

azhural, 2019-06-12:
> You should not have to build PCL because with ROS Melodic / Ubuntu 18.04 you get PCL 1.8.1 installed from the debians. (relevant to read: ros-industrial-consortium/bezier_examples#5)

libpcl-dev uses VTK6.3. Since bezier requires >7 you would still need to compile VTK and PCL from source.

DanielSchimpf, 2019-06-20:
Perfect that worked!
I compiled PCL 1.9.1 from source and afterwards bezier compiled successfully. Still have some warnings but I think I'll be able to resolve them. 
I think the issue can be closed now.

Thanks again!

<details>
  <summary>compilation log</summary>

```

dan@dan-VirtualBox:~/catkin_workspace$ catkin build
-----------------------------------------------------------------------
Profile:                     default
Extending:             [env] /home/dan/catkin_ws/devel:/opt/ros/melodic
Workspace:                   /home/dan/catkin_workspace
-----------------------------------------------------------------------
Build Space:        [exists] /home/dan/catkin_workspace/build
Devel Space:        [exists] /home/dan/catkin_workspace/devel
Install Space:      [unused] /home/dan/catkin_workspace/install
Log Space:         [missing] /home/dan/catkin_workspace/logs
Source Space:       [exists] /home/dan/catkin_workspace/src
DESTDIR:            [unused] None
-----------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
-----------------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-----------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-----------------------------------------------------------------------
Workspace configuration appears valid.

NOTE: Forcing CMake to run for each package.
-----------------------------------------------------------------------
[build] Found '1' packages in 0.0 seconds.                                     
[build] Updating package table.                                                
Starting  >>> catkin_tools_prebuild                                            
Finished  <<< catkin_tools_prebuild                [ 2.5 seconds ]             
Starting  >>> bezier                                                           
_______________________________________________________________________________
Warnings   << bezier:cmake /home/dan/catkin_workspace/logs/bezier/build.cmake.000.log
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
** WARNING ** io features related to libusb-1.0 will be disabled
cd /home/dan/catkin_workspace/build/bezier; catkin build --get-env bezier | catkin env -si  /usr/bin/cmake /home/dan/catkin_workspace/src/bezier --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/dan/catkin_workspace/devel/.private/bezier -DCMAKE_INSTALL_PREFIX=/home/dan/catkin_workspace/install; cd -
...............................................................................
Finished  <<< bezier                               [ 31.3 seconds ]            
[build] Summary: All 2 packages succeeded!                                     
[build]   Ignored:   None.                                                     
[build]   Warnings:  1 packages succeeded with warnings.                       
[build]   Abandoned: None.                                                     
[build]   Failed:    None.                                                     
[build] Runtime: 33.8 seconds total.                                           
[build] Note: Workspace packages have changed, please re-source setup files to use them.

 ```

</details>

VictorLamoine, 2019-06-21:
Warnings are from PCL: some modules were not compiled so are disabled. Nothing to worry about.
Thanks for the heads up

closed 2019-06-21 by VictorLamoine